blueman-project/blueman #1990: Double-clicking the device in blueman-manager sometimes doesn't connect it.

84634E1A607A, 2022-12-31:
blueman: blueman/vera,now 2.3.4+mint1+vanessa amd64
BlueZ: bluez/jammy,now 5.64-0ubuntu1 amd64
Distribution: Linux Mint 21.1 x86_64
Desktop environment: Cinnamon 5.6.5

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

*Sometimes* (Yes, only sometimes) when I open the blueman-manager and double-click my earphone, there is no response and right-clicking says "Connecting".

[image]

Then if I open the device menu, "Connect" is enabled, clicking it connects my earphone just fine.

[image]

Then I can double-click to disconnect it and double-click to connect again.

cschramm, 2022-12-31:
Could be #1905, fixed in 2.3.5. Disabling "Hide unnamed devices" in the "View" menu should avoid it then.

84634E1A607A, 2022-12-31:
I disabled it, but the bug still happens.

cschramm, 2022-12-31:
Ah, it's another bug that's fixed in 2.3.5, by https://github.com/blueman-project/blueman/commit/e6d051f52b27758f3f70e0bada5a7b64e3320b2b.

closed 2022-12-31 by cschramm