blueman-project/blueman #2003: Can not discover iclever keyboard bk10

DrStrom66, 2023-01-09:
blueman: 2.3.4
BlueZ: 5,64
Distribution: Linux Mint 21.1 Cinnamon
Desktop environment: Cinnamon

I can't discover an Iclever keyboard BK10.
but ii is discover-able by two Windows laptops

cschramm, 2023-01-09:
What name does Windows show when it discovers the device? Did you try disabling "Hide unnamed devices" in the "View" menu?

DrStrom66, 2023-01-12:
Hi Christopher,

Thanks for the tip with disabling unnamed devices, that made the
keyboard detectable. I have it now in the list.
But still I can't use it. It is marked as trusted and when I click on
connect I get following error : Connection Failed: br-connection-
create-socket

cschramm, 2023-01-12:
I would not expect the connection to succeed for an unpaired keyboard, but wouldn't expect the socket error either. You can check if `dmesg` or `journalctl -u bluetooth.service` have some info on it.

First thing you need to do is pairing. Typically when devices are pairable, they provide a name, but it might fall into #1954.

closed 2023-03-23 by github-actions[bot]