blueman-project/blueman #2005: blueman-applet error on startup

vinjana, 2023-01-10:
blueman: blueman/jammy,now 2.2.4-1
BlueZ: bluez/jammy,now 5.64-0ubuntu1
Distribution: Ubuntu 22.04.1 LTS
Desktop environment: LXQT (lxqt/jammy 30ubuntu1)

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

infirit, 2023-01-11:
Try removing `/var/lib/blueman/network.state` and try again.

cschramm, 2023-01-11:
If possible, please provide the `/var/lib/blueman/network.state` before removal. :smile: 

Did you update from an earlier Ubuntu release?

vinjana, 2023-01-11:
O.k., the file is empty. I anyway renamed it and restarted `blueman-applet`, and :tada: -- no error message anymore. Perfect! That was easy. Thanks!

Indeed, the system was updated from Ubuntu 20.04 to Ubuntu 22.04.

cschramm, 2023-01-11:
🤔 Surprising.

cschramm, 2023-01-11:
After a quick look at the code, I don't see many chances for something like this. The best explanation is probably that the `pickle.dump` call failed. If you still have the modification time of the file, you could check at least .xsession-errors for something related at that time.

In any case, that file is not used in the current code (upcoming 2.4) anymore. 🙂

vinjana, 2023-01-12:
Unfortunately, the error bugged me already some weeks and the `.xsession-errors` file for that time is lost.

cschramm, 2023-01-12:
I think we cannot find any traces then. Anyway, we dump the file in 2.4, so the issue is most probably not relevant anymore.

vinjana, 2023-01-12:
Thanks for you really swift support! Sorry that I couldn't provide you with more debugging information. At least from my side the issue could be closed.

cschramm, 2023-01-12:
> we dump the file

No pun intended. :laughing:

PR #2007 (2023-01-15): NetConf: Handle empty state files
fixes #2005
commit 2e7b4fee (2023-01-15): NetConf: Handle empty state files

fixes #2005
closed 2023-01-12 by cschramm