blueman-project/blueman #2012: Cannot pair mouse with blueman

WhyNotHugo, 2023-01-17:
blueman: 2.3.5-r0
BlueZ: 5.66-r2
Distribution: Alpine Linux
Desktop environment: swaywm

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I'm trying to pair a mouse with my desktop computer. Bluetooth is working, and I use bluetooth headsets on it that work just fine.

I put my mouse in pairing mode, and then click on "Search" in blueman. The mouse shows up and I click connect. Blueman shows a notification indicating that there is a pairing request and another indicating that the mouse is connected. However, the mouse continues to blink its light (e.g.: indicating that it is still in pairing mode) and does not work. After around a minute, blueman shows a final notification that the mouse has disconnected.

`blueman-manager` only logs this:

```
blueman-manager 17.22.18 WARNING  ManagerDeviceList:396 _monitor_power_levels: Failed to get power levels, probably a LE device.
```

If I try to pair the mouse via blueman, I get:

```
blueman-manager 17.17.58 ERROR    Manager:223 error_handler:  In Progress
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/blueman/bluez/Base.py", line 78, in callback
    value = proxy.call_finish(result).unpack()
            ^^^^^^^^^^^^^^^^^^^^^^^^^
gi.repository.GLib.GError: g-io-error-quark: GDBus.Error:org.bluez.Error.InProgress: In Progress (36)
blueman-manager 17.18.06 ERROR    Manager:223 error_handler:  Authentication Canceled
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/blueman/bluez/Base.py", line 78, in callback
    value = proxy.call_finish(result).unpack()
            ^^^^^^^^^^^^^^^^^^^^^^^^^
gi.repository.GLib.GError: g-io-error-quark: GDBus.Error:org.bluez.Error.AuthenticationCanceled: Authentication Canceled (36)
```

Nor `blueman-manager` nor `blueman-applet` seems to log any errors.

WhyNotHugo, 2023-01-17:
Oh, I see that `/var/log/messages` prints:

```
daemon.err bluetoothd[3488]: src/service.c:service_accept() input-hog profile accept failed for XX:XX:X:XX:XX:XX:XX
daemon.err bluetoothd[3488]: src/service.c:service_accept() input-hog profile accept failed for XX:XX:X:XX:XX:XX:XX
```

WhyNotHugo, 2023-01-17:
I can connect and pair to the device using `bluetoothctl`, but it still doesn't work.

WhyNotHugo, 2023-01-17:
Looks like the initial issue (not being able to connect/pair properly) is a blueman issue.

However, even if I use another tool, the mouse won't work. That's obviously not a blueman issue (I'll ask on the bluez tracker about that).

cschramm, 2023-01-17:
Please check if your notification daemon works as expected, see https://github.com/blueman-project/blueman/wiki/Troubleshooting#pairing-fails-on-the-remote-devices-side-wihout-a-pairing-dialog-showing.

WhyNotHugo, 2023-01-18:
Notifications are working fine, like I said, they show up fine:

> Blueman shows a notification indicating that there is a pairing request and another indicating that the mouse is connected.

The steps in the troubleshooting guide work fine, the notification renders properly:

[image]

cschramm, 2023-01-19:
> Notifications are working fine, like I said, they show up fine:
> 
> > Blueman shows a notification indicating that there is a pairing request and another indicating that the mouse is connected.

:facepalm: Reading is key.

> I don't think those are relevant here anyway.

They absolutely are as they are used to respond to pairing requests as well as service authorizations (which can usually be skipped by setting the device to trusted).

I would not expect `AuthenticationCanceled` in case blueman does not send a response (as there is none from the user), but I might be mistaken. Could you provide output from `blueman-applet --loglevel debug` of such a pairing request?

closed 2023-03-28 by github-actions[bot]