blueman-project/blueman #2014: Cannot mute/unmute Sandberg Bluetooth Office Headset Pro+

janstanka, 2023-01-18:
blueman: 2.3.4
BlueZ: 5.64-0ubuntu1
Distribution: Mint 21.1
Desktop environment: Cinnamon 5.6.7

In Linux Mint 21 Blueman replaces Blueberry. 
In my case I noticed malfunctional mic mute button on my Sandberg Bluetooth Office Headset Pro+ after upgrade to Mint 21. Mute button doesn't mute/unmute microphone anymore. With Blueberry in Mint 20 is OK.

infirit, 2023-01-18:
Blueman doesn't provide a mute button so I'm assuming you are referring to your desktop audio mixer. This isn't handled by blueman but by the audio server (usually pulseaudio) and mixer so my suggestion is to report this on the mint forums.

janstanka, 2023-01-19:
I mean mute button on headset - firstly it must be recognized by bluetooth manager (...I hope).

closed 2023-04-05 by github-actions[bot]