blueman-project/blueman #2015: cant cant to any device through bluetooth,bluetooth not working

badalraj09, 2023-01-19:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

wlarch, 2023-01-20:
I am experiencing the same issue since the last update. 
I cannot find any devices nor connect to previously paired devices.

blueman : 2.3.5

Distributor ID:	Linuxmint
Description:	Linux Mint 21.1
Release:	21.1
Codename:	vera

Desktop env. : cinnamon 5.6.7

cschramm, 2023-01-22:
@wlarch: What exactly do you mean with "last update"? Are you reporting a regression in blueman 2.3.5 (vs. 2.3.4)? So your system did not change otherwise and does find devices, showing e.g. in bluetoothctl but not in blueman?

wlarch, 2023-01-24:
@cschramm Thanks for the follow-up. It seems the problem has been resolved after a system reboot. I will get back here with more details if the problem repeats.

closed 2023-01-27 by infirit