blueman-project/blueman #2016: Device Tab Not Opening

Nick-Miras, 2023-01-21:
blueman: 2.3.5
BlueZ: 5.66
Distribution: Gentoo
Desktop environment: LXQT

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2023-01-21:
Quick info for directions, but I'll get back to this later: `socket(AF_BLUETOOTH, SOCK_RAW, BTPROTO_RFCOMM)` fails and `errno` should be set (blueman does not read it).

cschramm, 2023-01-22:
This should show you the immediate reason:

```sh
echo '#include <bluetooth/bluetooth.h>
#include <errno.h>
#include <stdio.h>
#include <string.h>
#include <sys/socket.h>

int main(void) {
    socket(AF_BLUETOOTH, SOCK_RAW, BTPROTO_RFCOMM);
    printf("%s (%d)\n", strerror(errno), errno);
    return 0;
}' >test.c

gcc test.c
./a.out
```

My guess is that the kernel just lacks RFCOMM support (`CONFIG_BT_RFCOMM`). I think bluetoothd generally does work with this (it just fails to connect affected devices), blueman-manager does not. We can try to make it work (basically just return an empty list from `SerialService.connected_instances` when `rfcomm_list` does not work; should be all, I guess), but is there any reason behind such an exotic kernel configuration?

infirit, 2023-01-22:
> but is there any reason behind such an exotic kernel configuration?

Most likely oversight when configuring the kernel. Gentoo does have a default kernel but a lot of people (like myself) build their own and forget to include important things.

Nick-Miras, 2023-01-24:
Thanks! It now works. Great oversight on my part on not realizing `CONFIG_BT_RFCOMM` was disabled.

infirit, 2023-01-26:
I'd like to keep this open as we shouldn't crash and burn like this. At the minimum we should handle this gracefully even if this unlikely to happen for most people.

closed 2023-10-29 by cschramm