blueman-project/blueman #2019: B450-XT II v1.05 headset not recognized as input device

jarrod843, 2023-01-26:
blueman: 2.3.2
BlueZ: 5.5.3
Distribution: Ubuntu 22.10
Desktop environment: Ubuntu Desktop

Blueman recognizes my headset , displays the correct model, and works for output, but does not seem to allow the system to select it as an input device. 

[image]

[image]

* [X ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

infirit, 2023-01-26:
Unfortunately pulseaudio doesn't support bluetooth HSP/HFP which makes the microphone work. If you want to use the microphone the only option at the moment is to use HSP/HFP through ofono. If you need help with that the ubuntu forums is probably the best place to ask.

cschramm, 2023-01-27:
To my understanding, ofono is not strictly required since pulseaudio 15 and Ubuntu 22.04 has something 16-ish, but in any case you have to switch the profile. Your sound settings how that it's set to A2DP which is just playback.

infirit, 2023-01-27:
didn't know. Good. Hopefully changing profile in pavucontrol works for you @jarrod843

closed 2023-04-06 by github-actions[bot]