blueman-project/blueman #2024: Support Flatpak.

rokejulianlockhart, 2023-02-04:
My rationale is available at https://github.com/safing/portmaster-packaging/issues/43#issue-956312378#:~:text=It%20shall%20allow,to%20support%20this.

In case this has already been completed, please know that https://flathub.org/apps/search/blueman and https://beta.flathub.org/apps/search/blueman provide no relevant results.

I am thankful for any assistance.

cschramm, 2023-02-05:
I have no experience with Flatpak but could imagine the sandboxing being quite a hassle for blueman.

In https://github.com/safing/portmaster-packaging/issues/45#issue-961633238 you seem to argue for Snap packaging instead. Why would you still use Flatpak then? :thinking:

rokejulianlockhart, 2023-02-06:
@cschramm, both are advantageous and (potentially) disadvantageous in different ways:

| which is superior: 	| why:                                                                                                                                                                                                                                                             	|
|--------------------	|------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------	|
| snap > flatpak     	| Snap supports adding a binary to $PATH to be invoked as <br>`example`, whereas Flatpak necessitates `flatpak run com.example.example`                                                                                                                            	|
| snap >= flatpak    	| Regarding confinement, both allow their sandbox to be significantly defeated: [`snap` uses `--classic`](https://snapcraft.io/docs/snap-confinement), whereas `flatpak` allows assigning a specific DBus communication permissions to use `flatpak spwan --host`. 	|
| flatpak > snap     	| Whereas the upstream `snap` source code solely supports one, central repository, which Canonical manages, `flatpak` is like any other package manager; it supports any that expose their packages in a standardized format.                                      	|

infirit, 2023-02-06:
I personally have zero interest in flatpak or any vendored application package format. If someone want to make one maintain it on flathub go for it.

rokejulianlockhart, 2023-02-06:
https://discourse.flathub.org/t/3707

closed 2023-04-15 by github-actions[bot]

rokejulianlockhart, 2023-04-17:
Please reopen.

cschramm, 2023-04-18:
I'm not personally interested in packaging blueman for Flatpak either, just like I'm not packaging blueman for any specific distribution or anything (except for the Debian package whose maintainer I happen to be). As neither of @infirit and me plans to work on this and to my understanding it does not have to be done upstream anyway (*), I don't see a value in having an open issue for it. What are your expectations?

(*) There's a lot of different packaging happening completely independent for distributions and other ecosystems like nixpksg, see https://repology.org/project/blueman/versions.

rokejulianlockhart, 2023-04-18:
@cschramm, that might work for me. I've not yet used Nix, but it seems interesting. However, without at least `flatpak`, you're missing-out on basically every non-technical user, and users of immutable OSes such as SteamOS (the Steam Deck's default) or Fedora Silverblue/Kinoite can't install it.

cschramm, 2023-04-19:
SteamOS 3 seems to be a good example to proof your point. :ok_hand: From what I read its default package management is indeed Flatpak (still, it's based on Arch and after "activating" its standard package manager one could just `pacman -Sy blueman`). I agree that it would be nice to have a blueman package at Flathub or whatever Flatpak repositories there are and make sense.

Anyway, I still don't see a value in keeping this issue open. I'm rather sure that communities like Flatpak / Flathub / SteamOS / whatever have their own trackers for packaging requests (seems like the Discourse link you posted actually is exactly that) and I cannot imagine anybody willing to build a Flatpak package to skim through our open issues and stumble upon the fact that there might be demand for packaging blueman. Personally I might give it a try at some point but definitely not by coming across a ticket drowning in other things that I once wanted to tackle. :sweat_smile: