blueman-project/blueman #2026: Bluetooth `Connection Failed: br-connection-unknown`

ovari, 2023-02-05:
From https://github.com/linuxmint/cinnamon/issues/11456

blueman: 2.3.5
BlueZ: not sure
Distribution: Linux Mint 21.1
Desktop environment: Cinnamon 5.6.7

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

### Issue
Unable to pair Linux Mint computer and Nokia phone.

### Steps to reproduce
1. Turn on Bluetooth
2. Pair Nokia phone with Linux Mint and `Connection Failed: br-connection-unknown`
[image]
4. Bluetooth seems to connect and then disconnect seconds later
[image]

### Expected behaviour
Able to pair Linux Mint and Nokia devices to test [Bug 128854 - LibreOffice Impress Remote for Android 2.4.0 and LibreOffice 6.3.3.2 for Linux](https://bugs.documentfoundation.org/show_bug.cgi?id=128854).

Thank you

cschramm, 2023-02-08:
Please provide the output from `blueman-applet --loglevel debug` of the pairing attempt as described on the troubleshooting page.

cschramm, 2023-02-09:
The process seems to terminate immediately. There's probably another blueman-applet process already running.

tai-tran-tan, 2023-02-18:
I had the same problem with keyboard Keychrone K8
Reproduce: 
1. connect keyboard for the first time (first time pair)
2. switch off keyboard
3. switch on keyboard, try to reconnect 
4. => notification shows 'connected' then 'disconnected' consecutively

Workaround: 
1. remove the keyboard from known device list (on laptop), 
2. reset keyboard to forget the laptop (on keyboard)
3. connect as first time pair => OK

Below are log entries from both failed and success connection (they are the same IMO)
Connection failed log:
`blueman-applet --loglevel debug
blueman-applet 11.36.14 INFO     PluginManager:87 load_plugin: ['PowerManager', 'NMPANSupport', 'DisconnectItems', 'AuthAgent', 'GameControllerWakelock', 'AutoConnect', 'ShowConnected', 'ConnectionNotifier', 'DBusService', 'StatusIcon', 'Menu', 'StandardItems', 'SerialManager', 'TransferService', 'Networking', 'ExitItem', 'NMDUNSupport', 'KillSwitch', 'DiscvManager', 'StatusNotifierItem', 'RecentConns', 'DhcpClient', 'NetUsage', 'PPPSupport']
blueman-applet 11.36.14 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 11.36.14 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 11.36.14 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Address': '50:E0:85:1E:A5:31', 'AddressType': 'public', 'Name': 'kali-linux', 'Alias': 'kali-linux', 'Class': 8126732, 'Powered': True, 'Discoverable': False, 'DiscoverableTimeout': 180, 'Pairable': True, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001133-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001', '0000110c-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb', '0000111e-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0541', 'Roles': ['central', 'peripheral']}
blueman-applet 11.36.14 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 11.36.14 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 11.36.14 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 11.36.14 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 11.36.14 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 11.36.14 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 11.36.14 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 11.36.14 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 11.36.14 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 11.36.14 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 11.36.14 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 11.36.14 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 11.36.14 INFO     Networking:35 load_nap_settings: Loading NAP settings
blueman-applet 11.36.14 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 11.36.14 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 11.36.14 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 11.36.14 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem'>
blueman-applet 11.36.14 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 11.36.14 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DhcpClient.DhcpClient'>
blueman-applet 11.36.14 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.PPPSupport.PPPSupport'>
blueman-applet 11.36.14 INFO     KillSwitch:106 io_event  : killswitch registered 1
blueman-applet 11.36.14 INFO     KillSwitch:122 io_event  : State: True
blueman-applet 11.36.14 INFO     PowerManager:179 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 11.36.14 INFO     KillSwitch:85 _on_connman_vanished: net.connman vanished
blueman-applet 11.36.14 INFO     TransferService:234 _on_dbus_name_appeared: org.bluez.obex :1.93
[... cut ...]
blueman-applet 11.38.08 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.PowerManager.PowerManager object at 0x7fcef40192a0>
blueman-applet 11.38.08 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <StatusIcon.StatusIcon object at 0x7fcef3ff5500 (blueman+plugins+applet+StatusIcon+StatusIcon at 0x5644f4969c40)>
blueman-applet 11.38.08 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.Menu.Menu object at 0x7fcef4019750>`

Hope this could help someone in investigation.
Cheers!

Susoiduts, 2023-02-22:
Hey, I am facing the almost the same issue as described by ovari with an bluetooth headset.

blueman: 2.3.5
BlueZ: 5.64
Distribution: Linux Mint 21.1
Desktop environment: Cinnamon 5.6.7

**Issue**
Unable to pair Linux Mint computer and SoundPeats Headset.

**Steps to reproduce**
Turn on Bluetooth
Pair SoundPeats headset with Linux Mint and `Connection Failed: br-connection-create-socket`

**Additional notes:**
- I tried the suggested reboot of the peripheral device as suggested by tai-tran-tan, but didn't work in my case. Issue stayed
- with other devices the bluetooth works fine (bluetooth speaker, smartphone)
- the headsets itself works also fine with other devices (smart phone, tablet other laptop with Linux Mint)

Here the log (I have run the log after reproducing the error):
```
blueman-applet --loglevel debug
blueman-applet 11.22.52 INFO     PluginManager:87 load_plugin: ['AuthAgent', 'ConnectionNotifier', 'AutoConnect', 'GameControllerWakelock', 'DiscvManager', 'KillSwitch', 'ExitItem', 'StatusNotifierItem', 'SerialManager', 'TransferService', 'ShowConnected', 'DBusService', 'Menu', 'NetUsage', 'StandardItems', 'PPPSupport', 'StatusIcon', 'NMDUNSupport', 'Networking', 'DhcpClient', 'PowerManager', 'DisconnectItems', 'NMPANSupport', 'RecentConns']
blueman-applet 11.22.52 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 11.22.52 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 11.22.52 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Address': '68:54:5A:B5:6F:D4', 'AddressType': 'public', 'Name': 'Machine', 'Alias': 'Machine', 'Class': 8126732, 'Powered': True, 'Discoverable': False, 'DiscoverableTimeout': 180, 'Pairable': True, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001133-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001', '00001108-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0540', 'Roles': ['central', 'peripheral']}
blueman-applet 11.22.52 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 11.22.52 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 11.22.52 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 11.22.52 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 11.22.52 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 11.22.52 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 11.22.52 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 11.22.52 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 11.22.52 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 11.22.52 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem'>
blueman-applet 11.22.52 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 11.22.52 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 11.22.52 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 11.22.52 WARNING  PluginManager:151 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 11.22.52 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 11.22.52 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 11.22.52 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 11.22.52 INFO     Networking:35 load_nap_settings: Loading NAP settings
blueman-applet 11.22.52 WARNING  PluginManager:151 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 11.22.52 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 11.22.52 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
[... cut ...]
blueman-applet 11.22.52 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.AuthAgent.AuthAgent object at 0x7f53c05dd030>
blueman-applet 11.22.52 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <StatusIcon.StatusIcon object at 0x7f53c19d4580 (blueman+plugins+applet+StatusIcon+StatusIcon at 0x55c141971f80)>
blueman-applet 11.22.52 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.Menu.Menu object at 0x7f53c05dd510>
```

cschramm, 2023-02-25:
`br-connection-create-socket` and `br-connection-unknown` are unexpected errors from BlueZ. You can probably find some information with `journalctl -u bluetooth.service`  after they happen, even without any debug logging.

Susoiduts, 2023-02-26:
Hey cschramm,

thank you for the hint! I did and got the output below.  Do you know what it means/how to proceed from there?

```
Feb 26 19:09:20 user bluetoothd[1033]: src/profile.c:ext_connect() Hands-Free Voice gateway failed connect to 22:88:33:08:23:16: Connection refused (111)
Feb 26 19:09:20 user bluetoothd[1033]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 22:88:33:08:23:16: Permission denied (13)
Feb 26 19:09:25 user bluetoothd[1033]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 22:88:33:08:23:16: Permission denied (13)
```

cschramm, 2023-02-26:
References on the Internets for that are typically for paired devices and the easy fix is to re-add them, sometimes in conjunction with power management. If the device is not paired, as you described, BlueZ might still have some state for it in `/var/lib/bluetooth` or something.

Susoiduts, 2023-02-26:
I tried readding several times but this doesn't change anything. What do you mean by conjunction with power management?
The pairing itself works (seems to) - the problem is the establishing a connection. Weird enough it also succeds occasionally and gets disconnect after 1-2minutes. While connected audio is played as it should.

cschramm, 2023-02-27:
I saw some references to power management, i.e. issues starting after standby or solved by changing some settings (nothing specific, though). The errors are often slightly different, though, e.g. `Permission denied (13)` in `ext_connect` and `Too many levels of symbolic links (40)` in `avdtp_connect_cb`.

The BlueZ project should definitely be more helpful to get behind the problem. One other thing than the remote device that I could think of causing the errors without knowing much of the kernel parts where they come from, is the audio server. Assuming you're using the pulseaudio default, you should check if its logs happen to give more insights and you might want to give PipeWire a shot.

DestyNova, 2023-03-21:
Like @tai-tran-tan I'm also having problems with a Keychron Bluetooth keyboard, but in my case a Keychron K6. My usual workaround is to switch the keyboard on and off several times until it automatically connects.

I never have connection problems with the other BT devices I'm using (a Logitech Pebble mouse and Soundcore Life Q30 headphones/headset), so I wonder if there's some quirk in the devices that experience issues.

alexey-stryi, 2023-03-29:
I have exact the same issue with the Keychron K8 pro keyboard as @tai-tran-tan described and it is very annoying.

cschramm, 2023-03-29:
There a bunch of hits on the Internets for Keychron keyboards and Linux Bluetooth. You might want to have a look at https://gist.github.com/andrebrait/961cefe730f4a2c41f57911e6195e444 and definitely try the `FastConnectable` option.

Moini, 2023-04-01:
Just had the same error message with my Soundcore Life Q30 headphones (for the first time). Removing the device from the list and then searching for it and connecting was enough to fix it.

wearysapper, 2023-08-20:
Just my two cents ...
Debian 12 , trying to connect Samsung earbuds using blueman-applet I got  "br-connection-adapter-not-powered"
this was with 6.4.0-0.deb12.2-amd64 kernel
but with 6.1.0-10-amd64 (and absolutely the same configuration)  connects like a charme automatically.
BTW i get still no sound because I don't want to use pulseaudio but alsa.
Hope this can help to identify the problem and solve the regression

gmjimenezc, 2023-08-24:
> Just had the same error message with my Soundcore Life Q30 headphones (for the first time). Removing the device from the list and then searching for it and connecting was enough to fix it.

This is what fixed it for me, restarted it using service instead of systemctl and then removed the device and searched for it again. I'm using kali linux.

gachiemchiep, 2023-09-12:
> Just had the same error message with my Soundcore Life Q30 headphones (for the first time). Removing the device from the list and then searching for it and connecting was enough to fix it.

Worked for me

kev-tsx, 2023-10-23:
> Just had the same error message with my Soundcore Life Q30 headphones (for the first time). Removing the device from the list and then searching for it and connecting was enough to fix it.

thanks, it worked for me too :)

Cyrilovich, 2023-10-26:
> Just had the same error message with my Soundcore Life Q30 headphones (for the first time). Removing the device from the list and then searching for it and connecting was enough to fix it.

Same here :) Thanks

sbirl, 2023-10-31:
I was also receiving `br-connection-unknown` for a first-time pair of my Soundcore Life Q20+ headphones to a Debian 12/bookworm.  Turns out my phone kept intercepting the connection which broke the pairing seconds after it said connected.
Didnt know about `--loglevel` (it's not in the `man` page), but this thread helped me locate the problem.  Thanks.

FallCheetah7373, 2023-11-06:
> Just had the same error message with my Soundcore Life Q30 headphones (for the first time). Removing the device from the list and then searching for it and connecting was enough to fix it.

works for me! I was using airdopes atom 81 and it worked without issues for 2 days and suddenly this issue started occuring out of nowhere repairing fixed it

not-fruit, 2023-11-27:
28/11/2023

I have a similar issue,
When running Linux Mint Vanessa the bluetooth device in question was able to connect. I have no trouble connecting this device to iPhone or Android.
Now running Victoria 21.2 only installed on 11 November and I had the issue on previous installs of the same release and on more than one machine since changing to this release about a year ago, With a fresh install on both machines, I thought I would have another go at getting it to work but there is no change.

Linux Mint 21.2 Victoria 64-bit 
Kernel Linux 5.15.0-89-generic x86_64
MATE 1.26.0
Bluetooth: Device-1: Intel Wireless-AC 9260 Bluetooth Adapter type: USB driver: btusb v: 0.8
   bus-ID: 1-11.1:5 chip-ID: 8087:0025
USB:Device-1: 1-11.1:5 info: Intel Wireless-AC 9260 Bluetooth Adapter type: Bluetooth
    driver: btusb rev: 2.0 speed: 12 Mb/s chip-ID: 8087:0025
slightly different hardware on second machine, same OS and kernel, same behavior.

Bluetooth:   Device-1: IMC Networks Bluetooth Radio type: USB driver: btusb v: 0.8 bus-ID: 6-11:4
    chip-ID: 13d3:3414
USB:    Device-2: 6-11:4 info: IMC Networks Bluetooth Radio type: Bluetooth driver: btusb rev: 2.1
    speed: 12 Mb/s chip-ID: 13d3:3414
  Hub-9: 7-0:1 info: Super-speed hub ports: 6 rev: 3.0 speed: 5 Gb/s chip-ID: 1d6b:0003

More information
[system information.txt](https://github.com/blueman-project/blueman/files/13478048/system.information.txt)
[hcitrace.txt](https://github.com/blueman-project/blueman/files/13477919/hcitrace.txt)
[journalctl.txt](https://github.com/blueman-project/blueman/files/13477947/journalctl.txt)
[syslog.txt](https://github.com/blueman-project/blueman/files/13477949/syslog.txt)
[xsession-errors.txt](https://github.com/blueman-project/blueman/files/13477953/xsession-errors.txt)

cschramm, 2023-12-12:
@goebbe: Clearly depends on how you use both tools. If you e.g. use the pair action on the device with both, there should not be a difference. Same for the connect action. If you take different steps, differences are expected, of course.

goebbe, 2023-12-12:
@cschramm 
I can pair both of my Bluetooth headphones using blueman. 
However, I cannot get a connection (via right click, connect).
After pairing, I can, however, connect successfully to both headphones via bluetuith. 
So I wonder where the differences are. Either bluetuith does something different from blueman OR there is something special about the way Linux Mint integrates blueman?  

Anyway, this is just a user reporting his observation.

Please let me know If I can provide additional information.

cschramm, 2023-12-12:
So from the exact same paired state an explicit connect attempt works in bluetuith but fails in blueman? What about `bluetoothctl`?

goebbe, 2023-12-12:
Thank you for your fast response!
Now this really feels odd. I cannot reproduce the described problem with blueman right now. 
During the last couple of days I have been testing several Bluetooth devices, all of them have been connected for the first time, and I am pretty sure I had the described connection problem with each of the devices, directly after pairing (for the first time). 
Is there something special when trying to connect for the first time?

I will observe the behaviour during the next weeks and will also try to connect other new devices, if I can get my hands on them. If the issue occurs again, I will report, here.

cschramm, 2023-12-12:
> Is there something special when trying to connect for the first time?

Not if the device is already paired and you do not change something like the trusted state in between.

FallCheetah7373, 2023-12-12:
let me interject, I have been using blueman, it seems that this issue occurs on random, at least for the user once a disconnect occurs right click and connect doesn't fix it, you have to unpair and repair.

FallCheetah7373, 2023-12-31:
> let me interject, I have been using blueman, it seems that this issue occurs on random, at least for the user once a disconnect occurs right click and connect doesn't fix it, you have to unpair and repair.

I was doing this regularly so I ran into another bug lmao, this time I figured that `blueman-manager` in terminal actually gives a error code. 

SO now I cannot unpair/remove the device maybe reboot will fix this. But the error code is:
`blueman-manager 11.52.22 ERROR    Base:85 callback  : Unhandled error for org.bluez.Adapter1.RemoveDevice
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/bluez/Base.py", line 78, in callback
    value = proxy.call_finish(result).unpack()
gi.repository.GLib.GError: g-io-error-quark: GDBus.Error:org.bluez.Error.DoesNotExist: Does Not Exist (36)`

cschramm, 2024-01-01:
@FallCheetah7373: Sounds like bluetoothd would provide a device that it does not actually know. :dizzy_face: Reproducible and same behavior in `bluetoothctl`? Removing state from `/var/lib/bluetooth` probably fixes the issue, but the BlueZ project might be interested in a bug report with the broken state.

FallCheetah7373, 2024-01-01:
> @FallCheetah7373: Sounds like bluetoothd would provide a device that it does not actually know. 😵 Reproducible and same behavior in `bluetoothctl`? Removing state from `/var/lib/bluetooth` probably fixes the issue, but the BlueZ project might be interested in a bug report with the broken state.

I rebooted and its not reproducible ever since.

bartus89, 2024-03-03:
Hello, I'm new to the github community. I have a problem with the BA07 bluetooth adapter. It works fine on Windows 10, finds devices and connects to them. But on my Linux laptop, Manjaro searches for devices correctly, but when I try to connect, I get br-connection-unknown. I tried a bluetooth speaker on my phone and both have the same problem

bartus89, 2024-03-03:
blueman-applet --loglevel debug
blueman-applet 18.31.38 INFO     PluginManager:87 load_plugin: ['SerialManager', 'RecentConns', 'AutoConnect', 'Networking', 'NMDUNSupport', 'ExitItem', 'PPPSupport', 'Menu', 'KillSwitch', 'DBusService', 'GameControllerWakelock', 'StandardItems', 'ShowConnected', 'AuthAgent', 'DiscvManager', 'DhcpClient', 'ConnectionNotifier', 'DisconnectItems', 'StatusIcon', 'NetUsage', 'NMPANSupport', 'PowerManager', 'StatusNotifierItem', 'TransferService']
blueman-applet 18.31.38 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 18.31.38 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 18.31.38 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Address': 'F4:4E:FC:D4:DF:D1', 'AddressType': 'public', 'Name': 'bartosz-aspire5750g', 'Alias': 'bartosz-aspire5750g', 'Class': 7078156, 'Powered': True, 'Discoverable': True, 'DiscoverableTimeout': 0, 'Pairable': True, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['0000110e-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000111e-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0548', 'Roles': ['central', 'peripheral'], 'Manufacturer': 992, 'Version': 12}
blueman-applet 18.31.38 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 18.31.38 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 18.31.38 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 18.31.38 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 18.31.38 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 18.31.38 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 18.31.38 INFO     Networking:35 load_nap_settings: Loading NAP settings
blueman-applet 18.31.38 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 18.31.38 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 18.31.38 WARNING  PluginManager:151 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 18.31.38 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 18.31.38 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 18.31.38 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 18.31.38 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 18.31.38 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 18.31.38 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 18.31.38 WARNING  PluginManager:151 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 18.31.38 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 18.31.38 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 18.31.38 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 18.31.38 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem'>
blueman-applet 18.31.38 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 18.31.38 INFO     KillSwitch:85 _on_connman_vanished: net.connman vanished
blueman-applet 18.31.38 INFO     KillSwitch:106 io_event  : killswitch registered 2
blueman-applet 18.31.38 INFO     KillSwitch:122 io_event  : State: True
blueman-applet 18.31.38 INFO     PowerManager:179 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 18.31.38 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.TransferService.TransferService object at 0x7fd8451450d0>
blueman-applet 18.31.38 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem object at 0x7fd8489f5a10>
blueman-applet 18.31.38 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.NMPANSupport.NMPANSupport object at 0x7fd841daffd0>
[... cut ...]
blueman-applet 18.31.38 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.DBusService.DBusService object at 0x7fd848995c10>
blueman-applet 18.31.38 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <StatusIcon.StatusIcon object at 0x7fd841daeb00 (blueman+plugins+applet+StatusIcon+StatusIcon at 0x55e3c3eb8940)>
blueman-applet 18.31.38 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.Menu.Menu object at 0x7fd8489f32d0>

bartus89, 2024-03-03:
[image]

hopeseekr, 2024-03-09:
It used to work, but now I can only pair my Bose SoundLink II if I put it into Discovery mode. Otherwise, I get this timeout every time. It's really annoying.