blueman-project/blueman #2027: Crashes caused by `cairo`

LamprosPitsillos, 2023-02-07:
blueman:
```
Name            : blueman
Version         : 2.3.5-1
Description     : GTK+ Bluetooth Manager
Architecture    : x86_64
Depends On      : bluez  bluez-libs  gtk3  libnm  python-cairo  python-gobject
Optional Deps   : dnsmasq: Network Access Point (NAP) support [installed]
                  iproute2: Network Access Point (NAP) support [installed]
                  networkmanager: Dial Up Networking (DUN) and Personal Area Networking (PAN) support [installed]
                  pulseaudio-bluetooth: audio devices support [installed]
Packager        : Antonio Rojas <arojas@archlinux.org>
Install Date    : Sat 26 Nov 2022 11:15:16 AM EET
```

BlueZ:
```
Name            : bluez
Version         : 5.66-1
Architecture    : x86_64
Depends On      : libical  dbus  glib2  alsa-lib  json-c
Required By     : blueman  bluez-runit  pulseaudio-bluetooth
Optional For    : networkmanager
Conflicts With  : obexd-client  obexd-server
Install Date    : Sat 12 Nov 2022 12:07:16 PM EET

```
Distribution: Artix Linux
Desktop environment: (Window Manager) Qtille

ERROR:
The error pops up in all the commands `blueman-*` 
```
Traceback (most recent call last):
  File "/usr/bin/blueman-adapters", line 17, in <module>
    from blueman.Functions import create_parser, create_logger, set_proc_title
  File "/usr/lib/python3.10/site-packages/blueman/Functions.py", line 37, in <module>
    import cairo
  File "/home/inferno/.local/lib/python3.10/site-packages/cairo/__init__.py", line 1, in <module>
    from ._cairo import *  # noqa: F401,F403
ImportError: /home/inferno/.local/lib/python3.10/site-packages/cairo/_cairo.cpython-310-x86_64-linux-gnu.so: undefined symbol: cairo_tee_surface_index
```

cschramm, 2023-02-07:
Your cairo libraries are broken. The Pycairo package installed to `/home/inferno/.local/lib/python3.10` does not match the `libcairo.so.2` that your system provides it with.

mor3dr3ad, 2023-03-27:
I have the same issue on arch linux, what would be the fix for that? I have tried reinstalling python-cairo on arch...