blueman-project/blueman #2029: Sorting new and unpaired devices at the top of the list

infirit, 2023-02-10:
In the past we could tell which devices were new and flagged them accordingly. It got removed because we couldn't tell anymore by what BlueZ send us over dbus. As the number of devices in the list increases it can get pretty bothersome trying to find the newly discovered devices so I want to always sort them at the top. 

In my listbox experiments I came up with a points system that works well enough but as someone without any CS or formal training I'm not sure this is a good way to handle this so feel free to be critical :-).

I came up with this which can be easily added to the current TreeView. Is this an ok solution or do you have other suggestions?

```python
def listbox_sort(self, row1, row2):
    # < 0 if row1 should be before row2, 0 if they are equal and > 0 otherwise.
    if row1.device is None or row2.device is None:
        return 0

    row1_rank = 0
    row2_rank = 0

    if row1.device["Paired"]:
        row1_rank += 10
    if row2.device["Paired"]:
        row2_rank += 10

    if row1.device["Alias"] > row2.device["Alias"]:
        row1_rank += 1
    elif row1.device["Alias"] < row2.device["Alias"]:
        row2_rank += 1

    if row1_rank > row2_rank:
        return 1
    elif row1_rank < row2_rank:
        return -1
    else:
        return 0
```

cschramm, 2023-02-25:
The rank lets the logic seem far more complex at first sight than it actually is. You could just compare a simple tuple key for `device`: `(device["Paired"], device["Alias"])`.

Instead of the branches that compare the keys (or ranks), a well-known replacement for Python 2's `cmp` is `(key1 > key2) - (key1 < key2)` but of course the branches are far more obvious than casting-based "calculations".

So I'd do something like this:

```python
key1 = (row1.device["Paired"], row1.device["Alias"])
key2 = (row2.device["Paired"], row2.device["Alias"])

if key1 > key2:
    return 1
elif key1 < key2:
    return -1
else:
    return 0
```

infirit, 2023-02-26:
I never (had to) think much about sorting but this makes sense. Thanks.

I also added the connected state in the sort. Below shows alphabetical; unpaired first, connected second then everything else.

[image]

```python
def listbox_sort(self, row1, row2):
    # < 0 if row1 should be before row2, 0 if they are equal and > 0 otherwise.
    if row1.device is None or row2.device is None:
        return 0

    key1 = (row1.device["Paired"], not row1.device["Connected"], row1.device["Alias"])
    key2 = (row2.device["Paired"], not row2.device["Connected"], row2.device["Alias"])

    if key1 > key2:
        return 1
    elif key1 < key2:
        return -1
    else:
        return 0
```

closed 2023-02-26 by infirit