blueman-project/blueman #2030: Icon Color Systray

galen1423, 2023-02-13:
Hi,

This is not a bug but a request.

I m using **Ubuntu 22.04** & **Blueman Applet 2.3.5**.
Would that be possible to change the icon color (blue for example ^_^) as soon as a device is connected ?

Thanks for this useful software.

cschramm, 2023-02-14:
Assuming you're not using symbolic icons (check `gsettings get org.blueman.general symbolic-status-icons`), our icons are [blueman-tray](https://github.com/blueman-project/blueman/blob/main/data/icons/hicolor/scalable/status/blueman-tray.svg) and [blueman-active](https://github.com/blueman-project/blueman/blob/main/data/icons/hicolor/scalable/status/blueman-active.svg) but your system probably overrides them with something monochrome.

galen1423, 2023-02-14:
Thanks for your fast reply. After checking the status of the icons, it turns out it was set to 'False'. I set to 'True' and it works as expected with my icon theme (Kora)

closed 2023-02-14 by galen1423