blueman-project/blueman #2033: Newly paired BT headsets set to LegacyPairing, cannot pair because pin required

CodyTVWeber, 2023-02-24:
blueman: 2.2.4-1
BlueZ: 5.64-0ubuntu1pop1~1674313994~22.04~579884f
Distribution: Pop!_OS 22.04 jammy (x86-64)
Desktop environment: Cinnamon 5.2.7

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

CodyTVWeber, 2023-02-24:
[image]

CodyTVWeber, 2023-02-24:
If I switch my DE to Pop, it works without the PIN.  But if I switch my DE back to Cinnamon, it requires a pin for new device.

cschramm, 2023-02-25:
I assume by "Pop" you mean Gnome, Pop's default DE, and thus use gnome-bluetooth instead of blueman there.

You should be able to see the agent requests from BlueZ to either blueman or gnome-bluetooth with something like
```
sudo dbus-monitor --system interface=org.bluez.Agent1
```
I'd expect no difference there, but gnome-bluetooth might be doing something fancy in response. Filtering the replies is not that easy, but
```
sudo dbus-monitor --system interface=org.bluez.Agent1 type=method_return
```
should make *all* method returns visible along the method calls for the interface and the correct one's `reply_serial` will match the call's `serial`. Curious what gnome-bluetooth does.

infirit, 2023-03-04:
gnome-bluetooth uses connect and trust. Would be interesting to see if you pair and then connect in blueman or just click connect.

closed 2023-05-12 by github-actions[bot]