blueman-project/blueman #2043: connected with bluethooth device but no sound on bluethooth device

jakbin, 2023-03-13:
blueman: kali-rolling 2.3.5-2+b1 amd64
BlueZ: kali-rolling 5.66-1+kali1 amd64
Distribution: kali linux
Desktop environment: xfce4 4.18

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Logs : https://pastebin.com/qUM1KaPW

[image]

jakbin, 2023-03-13:
i updated bluez firmware and some other, and restarted my laptop, then i start bluetooth service.
then i got another error : https://pastebin.com/2dgCewcv

uic17bsc1008, 2023-03-25:
For `debian` user `kali` or `parrot os` especially. This will fix connection protocol error audio issue.
 
`sudo apt-get install pulseaudio`
create a blue.sh file add command give bellow to blue.sh file.
step - 1
`vi blue.sh`

step - 2
```
systemctl --user start pulseaudio.service
pulseaudio --start
```
run blue.sh file
`sh blue.sh`

It will fix all issue regarding audio issue or protocol issue or host issue.

cschramm, 2023-03-25:
The last link shows an incomplete backtrace of something I cannot even anticipate from the provided parts. I'd guess that some libraries are broken or incompatible so that bluetoothd crashes.

jakbin, 2023-04-23:
I have fix it by changing output source in pulseaudio audio mixer.

closed 2023-04-23 by jakbin