blueman-project/blueman #2048: Connection Failed: br-connection-page-timeout -- problem occurs after Mint 21 os upgrade

ForgoRew, 2023-03-24:
blueman: ?
BlueZ: ?
Distribution: Linux Mint 21 Cinnamon
Desktop environment: Linux Mint 21 Cinnamon

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Hey there! When I upgraded my OS to Linux Mint 21 Cinnamon, the Bluetooth pairing with my JBL TUNE500BT doesn't work.
I try it again and again and always the same report:
Connection Failed: br-connection-page-timeout

Do you know, where can be the problem?

Thank you a lot - it used to work on Linux Mint 20 Cinnamon, I don't know, why it started now.

And sorry for not filling the blueman and BlueZ fields, I don't know what it means...

cschramm, 2023-03-25:
Check `journalctl -u bluetooth.service`, `journalctl -u pulseaudio.service` and `dmesg` for hopefully more details on the error.

ForgoRew, 2023-03-25:
Thanks, the first command says this (the OS version 21 was installed on 23rd March at ~22:00):
```
-- Boot 8bdb552d447b45cca3d2f91726c600e8 --
Mar 22 18:03:12 xxxxx-Yoga-Slim-7-14ARE05 systemd[1]: Starting Bluetooth service...
Mar 22 18:03:12 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[860]: Bluetooth daemon 5.53
Mar 22 18:03:12 xxxxx-Yoga-Slim-7-14ARE05 systemd[1]: Started Bluetooth service.
Mar 22 18:03:12 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[860]: Starting SDP server
Mar 22 18:03:12 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[860]: Bluetooth management interface 1.21 initialized
Mar 22 18:03:12 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[860]: Failed to set mode: Blocked through rfkill (0x12)
Mar 22 18:03:13 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[860]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSink/sbc
Mar 22 18:03:13 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[860]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSource/sbc
Mar 22 18:03:27 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[860]: Endpoint unregistered: sender=:1.37 path=/MediaEndpoint/A2DPSink/sbc
Mar 22 18:03:27 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[860]: Endpoint unregistered: sender=:1.37 path=/MediaEndpoint/A2DPSource/sbc
Mar 22 18:03:28 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[860]: Endpoint registered: sender=:1.54 path=/MediaEndpoint/A2DPSink/sbc
Mar 22 18:03:28 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[860]: Endpoint registered: sender=:1.54 path=/MediaEndpoint/A2DPSource/sbc
Mar 22 18:03:34 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[860]: Failed to set mode: Blocked through rfkill (0x12)
Mar 22 19:34:18 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[860]: Terminating
Mar 22 19:34:18 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[860]: Endpoint unregistered: sender=:1.54 path=/MediaEndpoint/A2DPSink/sbc
Mar 22 19:34:18 xxxxx-Yoga-Slim-7-14ARE05 systemd[1]: Stopping Bluetooth service...
Mar 22 19:34:18 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[860]: Endpoint unregistered: sender=:1.54 path=/MediaEndpoint/A2DPSource/sbc
Mar 22 19:34:18 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[860]: Stopping SDP server
Mar 22 19:34:18 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[860]: Exit
Mar 22 19:34:18 xxxxx-Yoga-Slim-7-14ARE05 systemd[1]: bluetooth.service: Succeeded.
Mar 22 19:34:18 xxxxx-Yoga-Slim-7-14ARE05 systemd[1]: Stopped Bluetooth service.
-- Boot 6449757e371d4356ba20a4112181418f --
Mar 22 21:24:06 xxxxx-Yoga-Slim-7-14ARE05 systemd[1]: Starting Bluetooth service...
Mar 22 21:24:06 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[866]: Bluetooth daemon 5.53
Mar 22 21:24:06 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[866]: Starting SDP server
Mar 22 21:24:06 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[866]: Bluetooth management interface 1.21 initialized
Mar 22 21:24:06 xxxxx-Yoga-Slim-7-14ARE05 systemd[1]: Started Bluetooth service.
Mar 22 21:24:07 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[866]: Failed to set mode: Blocked through rfkill (0x12)
Mar 22 21:24:07 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[866]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSink/sbc
Mar 22 21:24:07 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[866]: Endpoint registered: sender=:1.37 path=/MediaEndpoint/A2DPSource/sbc
Mar 22 21:24:20 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[866]: Endpoint unregistered: sender=:1.37 path=/MediaEndpoint/A2DPSink/sbc
Mar 22 21:24:20 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[866]: Endpoint unregistered: sender=:1.37 path=/MediaEndpoint/A2DPSource/sbc
Mar 22 21:24:20 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[866]: Endpoint registered: sender=:1.54 path=/MediaEndpoint/A2DPSink/sbc
Mar 22 21:24:20 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[866]: Endpoint registered: sender=:1.54 path=/MediaEndpoint/A2DPSource/sbc
Mar 22 21:24:27 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[866]: Failed to set mode: Blocked through rfkill (0x12)
Mar 22 23:09:07 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[866]: Terminating
Mar 22 23:09:07 xxxxx-Yoga-Slim-7-14ARE05 systemd[1]: Stopping Bluetooth service...
Mar 22 23:09:07 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[866]: Endpoint unregistered: sender=:1.54 path=/MediaEndpoint/A2DPSink/sbc
Mar 22 23:09:07 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[866]: Endpoint unregistered: sender=:1.54 path=/MediaEndpoint/A2DPSource/sbc
Mar 22 23:09:07 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[866]: Stopping SDP server
Mar 22 23:09:07 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[866]: Exit
Mar 22 23:09:07 xxxxx-Yoga-Slim-7-14ARE05 systemd[1]: bluetooth.service: Deactivated successfully.
Mar 22 23:09:07 xxxxx-Yoga-Slim-7-14ARE05 systemd[1]: Stopped Bluetooth service.
Mar 22 23:09:07 xxxxx-Yoga-Slim-7-14ARE05 systemd[1]: Starting Bluetooth service...
Mar 22 23:09:07 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[676212]: Bluetooth daemon 5.64
Mar 22 23:09:07 xxxxx-Yoga-Slim-7-14ARE05 systemd[1]: Started Bluetooth service.
Mar 22 23:09:07 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[676212]: Starting SDP server
Mar 22 23:09:07 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[676212]: Bluetooth management interface 1.21 initialized
Mar 22 23:09:07 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[676212]: Failed to set mode: Blocked through rfkill (0x12)
Mar 22 23:09:07 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[676212]: Endpoint registered: sender=:1.54 path=/MediaEndpoint/A2DPSink/sbc
Mar 22 23:09:07 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[676212]: Endpoint registered: sender=:1.54 path=/MediaEndpoint/A2DPSource/sbc
Mar 22 23:24:27 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[676212]: Endpoint unregistered: sender=:1.54 path=/MediaEndpoint/A2DPSink/sbc
[... cut ...]
Mar 25 10:08:19 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[754]: Endpoint registered: sender=:1.64 path=/MediaEndpoint/A2DPSink/sbc_xq_512
Mar 25 10:08:19 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[754]: Endpoint registered: sender=:1.64 path=/MediaEndpoint/A2DPSource/sbc_xq_512
Mar 25 10:08:19 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[754]: Endpoint registered: sender=:1.64 path=/MediaEndpoint/A2DPSink/sbc_xq_552
Mar 25 10:08:19 xxxxx-Yoga-Slim-7-14ARE05 bluetoothd[754]: Endpoint registered: sender=:1.64 path=/MediaEndpoint/A2DPSource/sbc_xq_552
```

infirit, 2023-03-25:
`Failed to set mode: Failed (0x03)` seems to be a common issue with some bluetooth chips. Which bluetooth adapter do you have (lsusb or lspci should show it).

uic17bsc1008, 2023-03-25:
For debian user kali or parrot os especially. This will fix connection protocol error audio issue.
 
`sudo apt-get install pulseaudio`
create a blue.sh file add command give bellow to blue.sh file.
step - 1
`vi blue.sh`

step - 2
```
systemctl --user start pulseaudio.service
pulseaudio --start
```
run blue.sh file
`sh blue.sh`

It will fix all issue regarding audio issue or protocol issue or host issue.

ForgoRew, 2023-03-25:
@infirit 

The lsusb command had this output:
```
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 8087:0029 Intel Corp. AX200 Bluetooth
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 04f2:b6cb Chicony Electronics Co., Ltd Integrated Camera
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

The command lspci was more wordy, but didn't show anything with Bluetooth...

uic17bsc1008, 2023-03-25:
@infirit what did you use in it?

ForgoRew, 2023-03-25:
@uic17bsc1008 

hmm, it seems not working for me :/

Actually, I already had the `pulseaudio` command on my computer, however, I tried the commands anyways. When I tried to connect my headphones, it didn't work and showed the same message: 	`Connection Failed: br-connection-page-timeout`.

Actually you know what's funny? Sometimes it shows me briefly, that the headphones are connected, but it is followed by the mentioned error message shortly.

uic17bsc1008, 2023-03-25:
@infirit first kill the process then start pulseaudio using bellow command given
`pulseaudio -k`
`pulseaudio --start`

uic17bsc1008, 2023-03-25:
@infirit  I can fix your problem once i see properly.

uic17bsc1008, 2023-03-25:
@infirit  Please use this command as well.
`systemctl --user enable pulseaudio`
`systemctl --user start pulseaudio`

uic17bsc1008, 2023-03-25:
@infirit  visit on this link it might help you a lot.
https://unix.stackexchange.com/questions/703088/bluetooth-headset-unable-to-connect-org-bluez-error-failed-br-connection-profil

ForgoRew, 2023-03-25:
Thanks for advices.

Btw., command used by one guy on stackexchange.com page, which you mentioned, used command systemctl with status. It gave me this output:
```
~$ systemctl --user status pulseaudio
● pulseaudio.service - Sound Service
     Loaded: loaded (/usr/lib/systemd/user/pulseaudio.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2023-03-25 19:43:00 CET; 2min 47s ago
TriggeredBy: ● pulseaudio.socket
   Main PID: 9634 (pulseaudio)
      Tasks: 5 (limit: 8598)
     Memory: 7.0M
        CPU: 89ms
     CGroup: /user.slice/user-1000.slice/user@1000.service/session.slice/pulseaudio.service
             └─9634 /usr/bin/pulseaudio --daemonize=no --log-target=journal

Mar 25 19:43:00 xxxxx-Yoga-Slim-7-14ARE05 systemd[1134]: Starting Sound Service...
Mar 25 19:43:00 xxxxx-Yoga-Slim-7-14ARE05 systemd[1134]: Started Sound Service.
Mar 25 19:43:00 xxxxx-Yoga-Slim-7-14ARE05 pulseaudio[9634]: Could not find org.bluez.BatteryProviderManager1.RegisterBatteryProvider(), is bluetoothd started with experimental features enabled (-E flag)?
```

ForgoRew, 2023-03-28:
Maybe I can downgrade to the version which worked for me?

closed 2023-06-04 by github-actions[bot]

AlexanderMakarov, 2024-05-25:
Is there any solution for this issue? Got the same today :(

AlexanderMakarov, 2024-05-25:
BTW passed this issue with advice from https://forum.manjaro.org/t/bluez-battery-provider/85220/2, but now getting another error:
```
... bluetoothd[9287]: src/adv_monitor.c:btd_adv_monitor_power_down() Unexpected NULL btd_adv>
... bluetoothd[9287]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP r>
```