blueman-project/blueman #2052: Bluetooth does not find devices

Murad103627, 2023-03-25:
blueman: 2.3.5+mint1+vera
BlueZ: 5.64-0ubuntu1
Distribution: linux mint cinnamon 21.1
Desktop environment: linux

With bluetooth everything is in order, but when I look for devices it does not show anything. I want to connect bluetooth headphones, but they are not displayed. I am new to Linux, but I would like to solve the problem.
root: sudo systemctl status bluetooth.service
● bluetooth.service - Bluetooth service
     Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor pre>
     Active: active (running) since Sat 2023-03-25 14:47:48 EDT; 17min ago
       Docs: man:bluetoothd(8)
   Main PID: 8193 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 6775)
     Memory: 788.0K
        CPU: 110ms
     CGroup: /system.slice/bluetooth.service
             └─8193 /usr/lib/bluetooth/bluetoothd

мар 25 14:47:48 SVF1532G4RB bluetoothd[8193]: Starting SDP server
мар 25 14:47:48 SVF1532G4RB bluetoothd[8193]: Bluetooth management interface 1.>
мар 25 14:48:20 SVF1532G4RB bluetoothd[8193]: Player registered: sender=:1.38 p>
мар 25 14:48:20 SVF1532G4RB bluetoothd[8193]: Endpoint registered: sender=:1.38>
мар 25 14:48:20 SVF1532G4RB bluetoothd[8193]: Endpoint registered: sender=:1.38>
мар 25 14:48:20 SVF1532G4RB bluetoothd[8193]: Endpoint registered: sender=:1.38>
мар 25 14:48:20 SVF1532G4RB bluetoothd[8193]: Endpoint registered: sender=:1.38>
мар 25 14:48:20 SVF1532G4RB bluetoothd[8193]: Endpoint registered: sender=:1.38>
мар 25 14:48:20 SVF1532G4RB bluetoothd[8193]: Endpoint registered: sender=:1.38>
мар 25 14:48:20 SVF1532G4RB bluetoothd[8193]: Endpoint registered: sender=:1.38>
lines 1-22/22 (END)

infirit, 2023-03-28:
Follow the debugging bluemans steps and run blueman-applet from a terminal. Then start search for d vices and post the output here. https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging-blueman

inklu, 2023-04-04:
I've the same issue. Here is the debug info of blueman-applet.

blueman-applet 10.48.06 INFO     PluginManager:87 load_plugin: ['AuthAgent', 'AutoConnect', 'ConnectionNotifier', 'DBusService', 'DhcpClient', 'DisconnectItems', 'DiscvManager', 'ExitItem', 'GameControllerWakelock', 'KillSwitch', 'Menu', 'NMDUNSupport', 'NMPANSupport', 'NetUsage', 'Networking', 'PPPSupport', 'PowerManager', 'RecentConns', 'SerialManager', 'ShowConnected', 'StandardItems', 'StatusIcon', 'StatusNotifierItem', 'TransferService']
blueman-applet 10.48.06 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 10.48.06 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 10.48.06 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Address': '9C:2F:9D:91:EE:EC', 'AddressType': 'public', 'Name': 'manjklyule', 'Alias': 'manjklyule', 'Class': 8126732, 'Powered': True, 'Discoverable': False, 'DiscoverableTimeout': 0, 'Pairable': False, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001133-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001', '00001108-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0542', 'Roles': ['central', 'peripheral']}
blueman-applet 10.48.06 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 10.48.06 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 10.48.06 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 10.48.06 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 10.48.06 WARNING  PluginManager:151 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 10.48.06 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 10.48.06 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 10.48.06 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 10.48.06 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 10.48.06 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 10.48.06 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 10.48.06 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 10.48.06 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 10.48.06 WARNING  PluginManager:151 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 10.48.06 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 10.48.06 INFO     Networking:35 load_nap_settings: Loading NAP settings
blueman-applet 10.48.06 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 10.48.06 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 10.48.06 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 10.48.06 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 10.48.06 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem'>
blueman-applet 10.48.06 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 10.48.06 INFO     KillSwitch:106 io_event  : killswitch registered 0
blueman-applet 10.48.06 INFO     KillSwitch:122 io_event  : State: True
blueman-applet 10.48.06 INFO     PowerManager:179 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 10.48.06 INFO     KillSwitch:85 _on_connman_vanished: net.connman vanished
blueman-applet 10.48.06 INFO     KillSwitch:106 io_event  : killswitch registered 2
blueman-applet 10.48.06 INFO     KillSwitch:122 io_event  : State: True
blueman-applet 10.48.06 INFO     PowerManager:179 update_power_state: off False | foff False | on True | current state True | new state True
[... cut ...]
blueman-applet 10.48.06 INFO     Networking:71 set_nap   : set nap False
blueman-applet 10.48.06 INFO     TransferService:234 _on_dbus_name_appeared: org.bluez.obex :1.65
blueman-applet 10.48.06 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Pairable': True}
blueman-applet 10.48.06 DEBUG    DiscvManager:92 on_adapter_property_changed: prop Pairable True
blueman-applet 10.48.06 INFO     ShowConnected:50 enumerate_connections: Found 0 existing connections
blueman-applet 10.48.06 INFO     AgentManager:17 on_registered: /org/bluez/obex/agent/blueman

Akashlatya, 2023-04-04:
Sap driver initialization failed.

what i do for this in parrot os for connecting bluetooth

closed 2023-06-12 by github-actions[bot]