blueman-project/blueman #2056: blueman-applet crashed with gi.repository.GLib.GError in call_finish(): g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.InvalidArgs: Unknown action ‘Activate’ (16)

S-A-TOP, 2023-04-02:
blueman: blueman 2.3.5-2build1
BlueZ: 5.66-0ubuntu1
Distribution: Xubuntu Lunar Lobster
Desktop environment: Xfce

blueman-applet crashed with gi.repository.GLib.GError in call_finish(): g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.InvalidArgs: Unknown action ‘Activate’ (16)

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

infirit, 2023-04-02:
Can you provide debug log for the applet as described in https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging-blueman

cschramm, 2023-04-02:
@infirit: Isn't that what #1997 fixes?

S-A-TOP, 2023-04-02:
I only receive a notification of a crash on login from apport.

Not when i manually run blueman-applet in terminal.

$ blueman-applet --loglevel debug
blueman-applet 23.16.49 INFO     PluginManager:87 load_plugin: ['StatusNotifierItem', 'DisconnectItems', 'StandardItems', 'NMDUNSupport', 'NMPANSupport', 'DiscvManager', 'SerialManager', 'ExitItem', 'StatusIcon', 'PowerManager', 'TransferService', 'RecentConns', 'Menu', 'PPPSupport', 'AuthAgent', 'ShowConnected', 'DhcpClient', 'NetUsage', 'DBusService', 'Networking', 'GameControllerWakelock', 'KillSwitch', 'ConnectionNotifier', 'AutoConnect']
blueman-applet 23.16.49 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 23.16.49 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 23.16.49 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Address': '58:A0:23:68:D5:E3', 'AddressType': 'public', 'Name': 'SAT-HPE', 'Alias': 'SAT-HPE', 'Class': 268, 'Powered': True, 'Discoverable': False, 'DiscoverableTimeout': 180, 'Pairable': False, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001801-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0542', 'Roles': ['central', 'peripheral']}
blueman-applet 23.16.49 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem'>
blueman-applet 23.16.49 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 23.16.49 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 23.16.49 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 23.16.49 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 23.16.49 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 23.16.49 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 23.16.49 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 23.16.49 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.PPPSupport.PPPSupport'>
blueman-applet 23.16.49 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 23.16.49 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 23.16.49 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DhcpClient.DhcpClient'>
blueman-applet 23.16.49 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 23.16.49 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 23.16.49 INFO     Applet:63 _on_dbus_name_appeared: org.bluez :1.5
blueman-applet 23.16.49 INFO     Functions:114 launch    : Gtk eventtime is 0, not using LaunchContext
blueman-applet 23.16.49 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_30_53_C1_6B_BE_A0 {'Address': '30:53:C1:6B:BE:A0', 'AddressType': 'public', 'Name': 'WH-CH710N', 'Alias': 'WH-CH710N', 'Class': 2360324, 'Icon': 'audio-headset', 'Paired': True, 'Bonded': True, 'Trusted': True, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['00000000-0000-0000-0099-aabbccddeeff', '00001108-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '0000110d-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '0000111e-0000-1000-8000-00805f9b34fb'], 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False}
blueman-applet 23.16.49 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_FC_A8_9A_57_6D_C6 {'Address': 'FC:A8:9A:57:6D:C6', 'AddressType': 'public', 'Name': 'SRS-X11', 'Alias': 'SRS-X11', 'Class': 2360340, 'Icon': 'audio-card', 'Paired': True, 'Bonded': True, 'Trusted': True, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['00001108-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '0000110d-0000-1000-8000-00805f9b34fb', '0000111e-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb'], 'Modalias': 'bluetooth:v0039p13BFd2105', 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False}
blueman-applet 23.16.49 INFO     BluezAgent:56 register_agent: Register Agent
blueman-applet 23.16.49 INFO     ShowConnected:50 enumerate_connections: Found 0 existing connections
blueman-applet 23.16.49 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Pairable': True}
blueman-applet 23.16.49 DEBUG    DiscvManager:92 on_adapter_property_changed: prop Pairable True
blueman-applet 23.16.49 INFO     TransferService:234 _on_dbus_name_appeared: org.bluez.obex :1.52
[... cut ...] ', '00001104-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001']
blueman-applet 23.16.50 INFO     PowerManager:140 request_power_state: Requesting True
blueman-applet 23.16.50 INFO     PowerManager:125 check     : callbacks done
blueman-applet 23.16.50 INFO     PowerManager:96 set_adapter_state: True
blueman-applet 23.16.50 INFO     PowerManager:179 update_power_state: off False | foff False | on True | current state True | new state True

cschramm, 2023-04-03:
Ah, well, of course Apport is the answer to why users even notice anything. So, yes, fixed by #1997.

closed 2023-04-03 by cschramm

infirit, 2023-04-03:
Yeah, just wanted to be sure it wasn't something else that was causing it on applet start.