blueman-project/blueman #2057: error: br-connection-profile-unavailable

S-A-TOP, 2023-04-02:
blueman:2.3.5-2build1
BlueZ: 5.66-0ubuntu1
Distribution: Xubuntu Lunar Lobster
Desktop environment: Xfce

error: br-connection-profile-unavailable

solution found online:

systemctl --user enable pulseaudio
systemctl --user start pulseaudio

should this be part of the installation process ?

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

infirit, 2023-04-02:
> should this be part of the installation process ?

What installation process are you referring to?

But it's not up to blueman to enable/start a soundserver like PulseAudio/Pipewire. This is up to the user/distro to configure. I'm also pretty sure ubuntu enables this by default for their users.

cschramm, 2023-04-02:
Ubuntu enables systemd services by default, so `systemctl --user enable pulseaudio` should not do anything. If `systemctl --user start pulseaudio` was necessary once, the `pulseaudio` package might have just been installed during the user session and need a re-login. However, it's a default package, so that's unexpected as well. If `systemctl --user start pulseaudio` is necessary in every session, the user session setup is broken somehow. Nothing blueman related.

S-A-TOP, 2023-04-02:
> > should this be part of the installation process ?
> 
> What installation process are you referring to?
> 
> But it's not up to blueman to enable/start a soundserver like PulseAudio/Pipewire. This is up to the user/distro to configure. I'm also pretty sure ubuntu enables this by default for their users.

It was not enabled by default for me by ubuntu.

S-A-TOP, 2023-04-02:
> Ubuntu enables systemd services by default, so `systemctl --user enable pulseaudio` should not do anything. If `systemctl --user start pulseaudio` was necessary once, the `pulseaudio` package might [...]

It might not be related to blueman but the error appears in blueman when i try to connect to a device and i do not know what the error message even means.

cschramm, 2023-04-03:
#1811 catches the error code and emits messages for it.

https://github.com/blueman-project/blueman/blob/76427887d49f504d9a6ec6d7c84147f7e056a158/blueman/gui/manager/ManagerDeviceMenu.py#L185-L187

closed 2023-04-03 by cschramm