blueman-project/blueman #2058: if self.Device['Paired']: TypeError: 'NoneType' object is not subscriptable

mammique, 2023-04-02:
blueman: 2.14
BlueZ: 5.55-3.1+rpt2
Distribution: Raspbian
Desktop environment: LXDE

Hello, when I tried to configure a device nothing happens, and I get this message if i launch the manager from the console:

```
blueman-manager version 2.1.4 starting
Traceback (most recent call last):
  File "/usr/bin/blueman-assistant", line 271, in <module>
    Assistant(args)
  File "/usr/bin/blueman-assistant", line 114, in __init__
    if self.Device['Paired']:
TypeError: 'NoneType' object is not subscriptable
```

Any idea where this comes from?

Thanks,

Camille.

infirit, 2023-04-02:
That's quite an old version of blueman, we removed the assistant a while ago. Anything in the assistant can be done in the device menu as well. If it doesn't then let us know what you are trying to do.

mammique, 2023-04-02:
OK, thanks for the reply. Is there a newer version available for Raspbian arm64? Or Debian might package might be OK if only Python.

Thank you,

Camille.

cschramm, 2023-04-02:
deb packages are typically built for specific Python versions and I expect your system's to be Python 3.9. You could have luck with https://launchpad.net/~pipewire-debian/+archive/ubuntu/pipewire-upstream and suite `impish` to get blueman 2.3. Otherwise my https://blueman.cschramm.eu/debian/ should have "nightlies".

closed 2023-06-11 by github-actions[bot]