blueman-project/blueman #2059: Missing Option To Turn Off Bluetooth Automatically

napperley, 2023-04-05:
blueman: 2.3.4
BlueZ: 5.64
Distribution: Linux Mint 21.1
Desktop environment: Cinnamon

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

No option is available in Blueman to automatically turn off Bluetooth at startup.

infirit, 2023-04-06:
You can update `/etc/bluetooth/main.conf` and set `AutoEnable=false` under `[Policy]`

napperley, 2023-04-08:
Updating a configuration file to disable Bluetooth isn't user friendly, and defeats the purpose of using Blueman in the first place.

cschramm, 2023-04-12:
I'd argue that a frontend to  `/etc/bluetooth/main.conf` is out of blueman 's scope. It is rather something for a system settings UI. I agree that it would be a cool feature but honestly... it means managing a text-based configuration file with a vast amount of really complex options, some of them hardly ever used and compatibility mess. I absolutely do not see us building anything like that with the resources we have.

closed 2023-06-20 by github-actions[bot]