blueman-project/blueman #2062: Wayland support?

fritzrehde, 2023-04-09:
blueman: 2.3.5
BlueZ: 5.66
Distribution: Arch Linux
Desktop environment: river

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Does `blueman-manager` support wayland? I tried opening it but nothing happens, opening from the command line just blocks.

infirit, 2023-04-09:
Using it right now on wayland. Blueman doesn't do anything that requires wayland, it's more the underlying toolkit Gtk3. Run it from a terminal, add ` --loglevel debug` to the command and post it here.

fritzrehde, 2023-04-09:
```
blueman-manager 22.09.29 INFO     PluginManager:87 load_plugin: ['PulseAudioProfile', 'Notes', 'Services', 'Info']
blueman-manager 22.09.29 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.manager.PulseAudioProfile.PulseAudioProfile'>
blueman-manager 22.09.29 INFO     PulseAudioUtils:193 pa_context_event: 1
blueman-manager 22.09.29 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.manager.Notes.Notes'>
blueman-manager 22.09.29 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.manager.Services.Services'>
blueman-manager 22.09.29 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.manager.Info.Info'>
blueman-manager 22.09.29 INFO     PulseAudioUtils:193 pa_context_event: 2
blueman-manager 22.09.29 INFO     PulseAudioUtils:193 pa_context_event: 3
blueman-manager 22.09.29 INFO     Manager:131 on_dbus_name_appeared: org.bluez :1.1
Blueman applet needs to be running
blueman-manager 22.09.54 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.manager.Info.Info object at 0x7fb9a9741870>
blueman-manager 22.09.54 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.manager.Services.Services object at 0x7fb9a9d97df0>
blueman-manager 22.09.54 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.manager.Notes.Notes object at 0x7fb9a9d94250>
blueman-manager 22.09.54 INFO     PulseAudioUtils:367 _on_delete: Destroying PulseAudioUtils instance
blueman-manager 22.09.54 INFO     PulseAudioUtils:193 pa_context_event: 6
blueman-manager 22.09.54 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.manager.PulseAudioProfile.PulseAudioProfile object at 0x7fb9a9d940d0>
```
Do I need to be running blueman-applet? If so, how? I'm using river with waybar, not sure if I even have an applet tray, nor do I want one. I've used blueman-manager on bspwm /xorg previously without any applet tray.

cschramm, 2023-04-10:
A blueman-applet process needs to be running, yes. blueman-manager could auto-start it but that does not seem to be supported by your system or installation. Just run `blueman-applet` to start it and keep it open while using blueman-manager. The applet process brings all the agent parts and stuff and blueman-manager could not do much more than display state without it. You typically keep it running to e.g. handle incoming connections at any time, but you may just stop it.

closed 2023-06-17 by github-actions[bot]