blueman-project/blueman #2063: Connection Failed:  Protocol not available

darshakkanani, 2023-04-12:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

rafa7404, 2023-04-15:
he hecho de todo debe ser problema de hardware es una laptop hp envy 360

cschramm, 2023-04-15:
"Protocol not available" is definitely not a hardware issue. Most probably you're just trying to connect an audio device without an appropriate sound server in place. Installing package `pulseaudio-module-bluetooth` could do the trick.

linux-noob1, 2023-04-30:
> "Protocol not available" is definitely not a hardware issue. Most probably you're just trying to connect an audio device without an appropriate sound server in place. Installing package `pulseaudio-module-bluetooth` could do the trick.

I have installed package  "pulseaudio-module-bluetooth" haven't fixed the issue, any other suggestions been trying for over than 2 days now

cschramm, 2023-04-30:
@linux-noob1: The designated way for Bluetooth audio support (if any) is distribution specific. Your distribution might use e.g. use a different audio server than pulseaudio, require extra steps to effectively enable the module (or a pulseaudio service in general), or just not include Bluetooth audio support at all.

linux-noob1, 2023-04-30:
@cschramm Thanks for the reply i appreciate it, am running "Parrot OS 5.2 (Electro Ara)" i triple checked the hardware am using it's working perfectly so it's something to do with software but still cannot determine which is it

cschramm, 2023-04-30:
Bluetooth audio is obviously out of scope for a penetration testing system. I actually fail to find which default audio server it's using, but the release notes of 5.2 lead me to suspect PipeWire. See e.g. https://wiki.debian.org/BluetoothUser/a2dp#PipeWire then if Parrot does not have any own resources.

linux-noob1, 2023-05-01:
@cschramm Somehow it magically did work i been reinstalling all the services tried to upgrade my parrot os and i got it to connect perfectly now, to be honest with the amount of things i tried am not 100% sure what have got it fixed but last thing i have tried is your advice : 

i did run this : sudo apt install libspa-0.2-bluetooth=0.3.65-2~bpo11+1

and then did reboot the machine, tried to pair and connect my headset in cli with "bleutoothctl" and it worked like magic.

hope this help other people out there, and thanks for the advice you had me searching on the pipewire and ended up trying that and it worked.

Thenathayalan, 2023-06-28:
For me I have installed the package " pulseaudio-module-bluetooth " and we need to start pulse audio manually by running the command " pulseaudio --start " and now try to connect your Bluetooth device I hope this works 🙂

diveloper53, 2023-08-24:
> For me I have installed the package " pulseaudio-module-bluetooth " and we need to start pulse audio manually by running the command " pulseaudio --start " and now try to connect your Bluetooth device I hope this works 🙂

Thank you! I have the same issue, and it helped me!

closed 2023-10-31 by github-actions[bot]