blueman-project/blueman #2065: Dynaudio Xeo2 bluetooth speakers incorrectly displayed with audio-headphones icon

AJPoulter, 2023-04-15:
blueman: 2.3.5-1
BlueZ: 5.66-1
Distribution:archlinux
Desktop environment: endeavouros

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

infirit, 2023-04-15:
> Icon: audio-headphones

This is what BlueZ (linux bluetooth subsystem) gives us so that what we use. We are not BlueZ, just a frontend using it.

cschramm, 2023-04-15:
>  Class: 0x00240418

And this is what the device itself reports.

Audio, Rendering, Audio/Video, Headphones

https://btprodspecificationrefs.blob.core.windows.net/assigned-numbers/Assigned%20Number%20Types/Assigned_Numbers.pdf#10

closed 2023-06-22 by github-actions[bot]