blueman-project/blueman #2066: Bluetooth is not searching/connecting devices

kha1mamatov, 2023-04-16:
blueman:2.3.5-2+b1
BlueZ:5.66
Distribution: Kali GNU/Linux Rolling 2023.1
Desktop environment:

I have a problem with bluetooth. I cannot open devices tab to search it opening adapters everytime. here is some logs...

bluetooth status:
`● bluetooth.service - Bluetooth service
     Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; preset: disabled)
     Active: active (running) since Sun 2023-04-16 16:11:02 +05; 10s ago
       Docs: man:bluetoothd(8)
   Main PID: 16056 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 8999)
     Memory: 980.0K
        CPU: 54ms
     CGroup: /system.slice/bluetooth.service
             └─16056 /usr/libexec/bluetooth/bluetoothd`

bluetooth adapter(RTL8821CE
[image]
):
Bus 001 Device 004: ID 0bda:b00e Realtek Semiconductor Corp. Bluetooth Radio

cschramm, 2023-04-22:
Do you mean the "Device" menu? It's for the currently selected device. As the list is empty, there can't be one. That's not where you start searching. Try the big search button instead.

In case you can't do anything useful, you might just not have a working adapter. As long as e.g. `bluetoothctl` doesn't show one, blueman can't do anything for you.

Norparn, 2023-04-23:
I have the exact same problem, after install i found an adaptor! But I didn't marked it before the system was rebooted. 
How remove and reactivate the bluetoth adapter?
The system says I do not need any new drivers!
/Nor

cschramm, 2023-04-24:
@Norparn: You do not have an adapter e.g. in `blueman` or `bluetoothctl` but you do see one e.g. with `lsusb`  or `lspci`? No driver picked it up properly then.

Norparn, 2023-04-24:
Tnx for answer, to day when I started my laptop I tried to connect to my
Bluetooth with my mouse. Behold they made contact and I make it permanent.
Reboot seems to solved the problem.
/ Nor

Shyamvalliattu, 2023-05-13:
I have the exact same problem in my laptop. Do anyone have a permanent solution to it ?

donottellmetonottellyou, 2023-05-20:
I am having this problem, using Linux Mint 21.1. Device is greyed out.

Manukula, 2023-06-10:
so how do u fix the bluetooth not connecting problem, should it be something you must download in linix mintor what

closed 2023-08-18 by github-actions[bot]