blueman-project/blueman #2068: Can't conect to smartphone for audio profile

blackshark1213, 2023-04-20:
blueman: install 
BlueZ: install 
Distribution: Kali GNU/Linux Rolling x86_64
Desktop environment: xfce 4.18

Connection Failed: br-connection-profile-unavailable 

while connecting to smartphone 
plz help  me to resolve this problem

cschramm, 2023-04-20:
Most probably you're trying to connect an audio device without an appropriate sound server in place. If you're using pulseaudio, installing package `pulseaudio-module-bluetooth` could do the trick.

marwank270, 2023-04-22:
Hi, I have the same problem since a month, same configuration but with a headseat and everything I tried didn't changed anything, tried to install `pulseaudio-module-bluetooth` and it's already in place

blackshark1213, 2023-04-24:
>

blackshark1213, 2023-04-24:
> Most probably you're trying to connect an audio device without an appropriate sound server in place. If you're using pulseaudio, installing package `pulseaudio-module-bluetooth` could do the trick.

pulseaudio-module-bluetooth 
Already it install in my system , but can't fix problem

closed 2023-04-24 by blackshark1213

cschramm, 2023-04-25:
You might be using a different sound server like PipeWire, your pulseaudio installation might have some issue and not work properly, etc. ...

Things to try include checking `systemctl status --user pulseaudio.service` and looking for media endpoint registrations (from the user's pulseaudio daemon) in `sudo journalctl -u bluetooth.service`.