blueman-project/blueman #2070: How do I disable the "Connected" window that shows up when I connect my Bluetooth headphones?

timakro, 2023-05-02:
blueman: 2.3.5
BlueZ: 5.66
Distribution: Arch
Desktop environment: none (plain openbox)

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

How do I disable this annoying window?

[image]

The ConnectionNotifier and AutoConnect plugins are off.

[image]

From a quick code search it seems to be triggered [here](https://github.com/blueman-project/blueman/blob/c7ec223c87ddce8b18b00776861e73aa6799b64d/blueman/plugins/applet/AutoConnect.py#L50).

cschramm, 2023-05-02:
The notification is from https://github.com/blueman-project/blueman/blob/c7ec223c87ddce8b18b00776861e73aa6799b64d/blueman/plugins/applet/RecentConns.py#L155

It's not possible to use the `RecentConns` plugins (i.e. connect to previously connected services from the applet menu) without that notification. You can connect from the devices window (`blueman-manager`) to avoid it.

Also note that you seem to see a fallback window for broken setups. blueman depends on a notification daemon (following the `Desktop Notifications Specification`). A typical choice for minimalist window managers seems to be `dunst`, but of course you could also use `xfce4-notifyd` or anything else. I'm pretty sure you can also configure some of them to just drop the notifications they receive. :see_no_evil: :speak_no_evil:

closed 2023-05-02 by cschramm

timakro, 2023-05-03:
Thanks for digging into this for me. I've tried `dunst` and it works great for me.