blueman-project/blueman #2075: Blueman does not find all my bluetooth devices

nico6293, 2023-05-11:
blueman: 2.3.5
BlueZ: ?
Distribution: Linux Mint 21 Cinnamon 5.4.12
Desktop environment: Lenovo G780 with usb bluetooth tp-link bluetooth 5.0

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Bluetooth finds my my new Galaxy A13 mobile phone.
Bluetooth does not find new bluetooth speaker Aukey SK-M28.
Bluetooth does not find Galaxy Buds2 pro wireless bluetooth earphones.

Both Aukey and Buds2 pro are found by Galaxy A13.

infirit, 2023-05-11:
Try disabling hiding unnamed devices in the view menu.

nico6293, 2023-05-11:
I found the problem. Thanks

closed 2023-05-11 by nico6293

Tom2320, 2023-09-29:
...houres later.... many thanks.