blueman-project/blueman #2079: blueman completely non functional

russ-burkert, 2023-05-17:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

russ-burkert, 2023-05-17:
what does 'not planned' mean?

cschramm, 2023-05-17:
You're welcome to provide at least a problem statement. There's nothing that we could help you with otherwise.

russ-burkert, 2023-05-17:
Hmmm...the title of this ticket seems like a clearly worded problem to me..."[blueman completely non functional].  When I run the Bluetooth Manager app, I should be able to search for visible bluetooth devices and pair with those devices.  The Search button on the Bluetooth Devices dialog finds no BT devices, even when I have bluetooth headphones inches away in pairing mode.  My original post shows the output from running the bluetoothctl app at the command line

cschramm, 2023-05-17:
Ha, it looks like you hid most of your original post in comment tags. I edited it to make it visible.

As you showed by unsuccessfully trying with `bluetoothctl`, your problem does not actually seem to have something to do with blueman.

You're clearly having a severe problem with your controller. I'd check `bluetooth.service` logs and `dmesg` for traces.

russ-burkert, 2023-05-17:
I looked at syslog and journalctl results and all I can tell is 'there is  problem'.  So, I'm going to suspend further attempts and go with an external USB dongle and see what happens.  I suspect my HP PC is just to old for these modern OSs

EspressoMan, 2023-05-21:
I'm also running Linux mint 21 on my 4 year old Lenovo Thinkpad. I have my late model pixel phone sitting beside my notebook. Bluetooth is enabled on both. Yet the laptop cannot discover the phone and the phone cannot see the laptop. This isn't necessarily a blueman issue but I don't recall experiencing this issue with previous versions of Linux Mint. @russ-burket isn't the only one experiencing this issue...

closed 2023-07-30 by github-actions[bot]