blueman-project/blueman #2080: apple magic trackpad keeps losing connection

ScherbakovAl, 2023-05-21:
Hello! Apple magic trackpad loses connection every few minutes. Communication is lost with ubuntu 23.04. For example, with my android phone, this trackpad works without any problems.
About this problem that exists and other people have been told here:
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1385113

This issue is most likely not related to the blueman software. But I thought that you, when developing software for working with bluetooth, are well versed in how it works. Maybe you could tell me what is the problem of frequent loss of bluetooth connection with magic trackpad in ubuntu 23.04?
I'll attach logs and information from hciconfig, bluetoothctl, output from syslog and sudo btmon:
https://pastebin.com/3A9EXusp

What other logs and data do I need to provide to help sort out this problem?
Where else can you report it?

Thank you!

luciferasee, 2023-06-01:
I'm have the same problem, my headphone lose the connection every moment

ScherbakovAl, 2023-06-02:
New error messages when communication is lost with Magic Trackpad:
I booted Ubuntu 23.04(kernel 6.2.0) from a flash drive - the same error:
https://pastebin.com/5KtUjCHQ
I booted from Fedor's(kernel 6.2.9) flash drive - error:
https://pastebin.com/4LJC7NJQ
When using Fedora, communication breaks also occur.

infirit, 2023-06-02:
None of this seems relevant to blueman? You may have confused us with BlueZ which is the project that actually handles the kernel drivers/low level userland. Blueman is just the frontend/.

ScherbakovAl, 2023-06-02:
> For example, with my android phone, this trackpad works without any problems. The problem does not go away when I use different usb dongles for bluetooth connection.

Thank you! I wrote to them!

ScherbakovAl, 2023-06-02:
> None of this seems relevant to blueman? You may have confused us with BlueZ which is the project that actually handles the kernel drivers/low level userland. Blueman is just the frontend/.

Thank you! I wrote to them!

closed 2023-08-09 by github-actions[bot]