blueman-project/blueman #2081: Gnome 44. Double blueman appindicator after resume from suspend

monte-monte, 2023-05-21:
blueman: 2.3.5
Distribution: After resuming my system from suspend blueman indicator in top panel is doubled.
Desktop environment: Gnome 44b

After upgrade from Ubuntu 22.10 to 23.04 the issue appeared. The only way to fix is to restart blueman.
I'm not sure if this connected to new version of Gnome or blueman, but I've found similar issues with blueman reported before, but seemingly fixed in newer versions. So maybe something similar here.

vageliskatsiotis, 2023-07-02:
Confirming this also on Arch x64, 6.1.37-1-lts, GNOME 44.3. Issue occurs also after restarting Gnome Shell via Alt-F2 -> r

cschramm, 2023-08-03:
I'd actually expect zero icons on Gnome. Are you using something like the "Tray Icons: Reloaded" extension?

monte-monte, 2023-08-04:
@cschramm that would be built-in "Ubuntu Appindicators" extension.

cschramm, 2023-08-06:
Looks like https://github.com/ubuntu/gnome-shell-extension-appindicator/issues/419 then.

monte-monte, 2023-08-06:
@cschramm Thanks a lot! Reverting https://github.com/ubuntu/gnome-shell-extension-appindicator/commit/41a8e9c7ff20c0d45ff91a6329455e6f31f2e4aa helped with the issue.
At first I didn't even realize that it wasn't even a native appindicator.

closed 2023-08-06 by monte-monte