blueman-project/blueman #2084: device connected but no audio can be heard

spiridonios, 2023-05-29:
Hi!  It's my first time posting on github and maybe it is a bit advanced for me, since i m' not a developer. My linux distribution help page directs me here though, so i will try asking and maybe my problem will be sorted out. 

I use this linux distribution for some years and i ve never been able to connect my laptop to any device via bluetooth. At the moment it looks like the laptop is connected to the amplifier, cause i read that i have a receival strength of 50%, a link quality of 100% and a transmit power level of 52%. The amp has also made a connection indicative sound and the bluetooth sign at the toolbar has turned green. But still, i listen through the laptop speakers, instead of the amplifier speakers. 

Would anyone have an advice on how to proceed in order to try and solve my issue?

blueman:
BlueZ:
Distribution:2.0.8
Desktop environment:linux Mx-Av

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

infirit, 2023-05-29:
Sounds like everything is connecting properly. On most linux distributions the audio sink doesn't automatically change when you connect a bluetooth speaker/amp. You need to change it with your desktop's audio mixer or something like pavucontrol.

This is how it looks like in plasma (Mi bluetooth speaker).
[image]

spiridonios, 2023-05-29:
thank you very much for the hint!
opening the mixer, i went to the output devices tab and enabled the Bluetooth port option. And just simple as that, i now have audio coming out of the speakers connected to the Bluetooth amp!

closed 2023-05-29 by spiridonios