blueman-project/blueman #2086: AssertionErrors when starting blueman

bulgariamitko, 2023-06-02:
blueman: 2.3.5
BlueZ: 5.66
Distribution: Ubuntu 23.04
Desktop environment: ubuntu:GNOME

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

infirit, 2023-06-02:
ugh, why is it making it a dbus method call 😞 . If you want to just get it working quickly remove this line @bulgariamitko https://github.com/blueman-project/blueman/blob/2-3-stable/blueman/gui/manager/ManagerDeviceList.py#L138

@cschramm you want to push out a release with a fix for this? It's not going to be an issue in main afaics.

infirit, 2023-06-02:
Wait, does blueman work even with these errors @bulgariamitko ?

bulgariamitko, 2023-06-02:
@infirit that is a good question. I don't know, but maybe the issue is on my end as sometimes and I do not know why my bluetooth just stop working and when I then try to start blueman-manager I get this issue, it is opening and then it ask me to enable bluetooth I do that and then I do not see anything, no devices. To fix this issue I have to turn-off my laptop and then unplug the power source from my com, wait for 10 secs plug it again and then it is working. Restarting will not fix my issue. 
So, like I said I do not know if this is a blueman issue, but maybe for cases like mine there need to be try/catch code?

cschramm, 2024-01-30:
Look like #1993 to me. The cause would be multiple competing faders, e.g. due to multiple calls to `clear`, wherever those come from. :shrug:

graue70, 2024-02-05:
@cschramm When can I expect a release that includes the fix? The fix has been in main for more than a year...