blueman-project/blueman #2088: Ubuntu 23, bluetooth works, then reboot, then bluetooth does not reconnect

hubbertsmith, 2023-06-06:
blueman:
BlueZ:
Distribution:
Desktop environment:

* configure for the first time, works fine
* reboot and will not reconnect
* i tried service bluetooth restart   
* I tried systemctl restart bluetooth

darrell-barabash, 2023-06-18:
I do believe that I am having the same issue. Ever since I upgraded to MINT 21 this same problem occurs on every computer running MINT 21. If I boot with MINT 20.3 no problem.

Even if previous connected devices are still listed and re-connect, it seems that the channel is broken. The only work around I have found is to 1) remove the problem device; 2) rescan for devices, and; 3) add the desired device back in again. Restarting bluetooth didn't work for me. Problem with this workaround is that if the computer is rebooted or even goes into standby, the same procedure has to be done again, and again ... 

Has anyone found a solution to this yet? I'd be more than happy do any further tests and provide data that might be helpful.

**TESTING I HAVE DONE:**
The bluetooth devices that I have been testing with is a couple Logitech M557 mice and several scuba diving computers. All work great on MINT 20.3.

I'm pretty sure this isn't a specific hardware issue. Three computers have the same issue with MINT 21 (blueman) whilst using different bluetooth HW:
- **Recent MSI Aegus PC with built-in wireless and BLE:** Bluetooth: Device-1: Intel type: USB driver: btusb v: 0.8 bus-ID: 1-14:7 chip-ID: 8087:0033 Report: hciconfig ID: hci0 rfk-id: 0 state: up
- **PC with a BLE dongle:** Device-3: 3-2:3 info: Cambridge Silicon Radio Bluetooth Dongle (HCI mode) type: Bluetooth driver: btusb rev: 2.0 chip ID: 0a12:0001
- **Laptop with built-in bluetooth adapter:** Bluetooth: Device-1: Broadcom HP Portable SoftSailing type: USB driver: btusb v: 0.8 bus-ID: 2-1.6:3 chip-ID: 0a5c:21e1 Report: hciconfig ID: hci0 rfk-id: 1 state: up address: <filter> bt-v: 2.1 lmp-v: 4.0 sub-v: 220e

darrell-barabash, 2023-06-19:
A bit more information that might be of help. My dive computer is a BLE device and was previously paired and trusted. During a session it stays connected with no issues. But, after a reboot or even just logging off, I see this message in the syslog where E1:52:44:09:D2:BF is my Shearwater dive computer. 

**Jun 18 10:36:53 beeker2 bluetoothd[1149]: src/device.c:load_gatt_db() No cache for E1:52:44:09:D2:BF**

If I remove the device, rescan and re-install all is good until the next logout/reboot. I didn't specifically get this info for the Logitech mice but the behavior is the same; have to uninstall and reinstall every time I logged in.

darrell-barabash, 2023-06-19:
**FIXED** -- even though this computer was a clean installation, I removed bluez and then reinstalled. Now it seems to keep all settings whether I logoff, switch users or reboot. This is especially convenient for the wireless mouse.

```
> sudo apt remove --purge bluez
> sudo apt install bluez
```

hubbertsmith, 2023-06-19:
thanks for the attention !!
I tried these exact steps ... it did not work for me.
if there is something I can do for repeatability, LMK
tnx

darrell-barabash, 2023-07-09:
I'm getting the impression that the issue is somehow related to blueman-applet. What I reported previously must have been a fluke because the problem eventually came back after a couple reboots. 

However, I think that I may have stumbled on something that does work. It has survived a number of reboots and logoffs and on two different MINT21 computers using a bluetooth mouse. What I did, and repeated on another computer with same issue, was:
1) In the blueman-applet, remove the mouse.
2) open bluetoothctl and type "devices"
3) if the mouse shows up there, "remove" it
4) Then, start pairing mode on the mouse
5) get the mouses Bluetooth Device Address by turning "scan on"
6) trust the mouse with "trust <Bluetooth Device Address>"
7) start pairing the mouse with "pair <Bluetooth Device Address>"
8) connect to to the mouse with "connect <Bluetooth Device Address>"
9) exit from bluetoothctl
9) finally, at CLI, and this might be a key step (???) for survivability:
    >> sudo systemctl enable --now bluetooth

Hopefully it will work for you. For me, 2 weeks and it is still working on two different computers so high confidence in this work-around.

cschramm, 2023-07-10:
To my understanding of `systemctl enable --now bluetooth` does not actually have any effect. It would enable (i.e. create some symlinks for) the bluetooth service, but if it's not enabled already you will not have Bluetooth at all after a reboot. Due to `--now` it would also start the service, but if it's not running already none of the previous steps would have been possible.

darrell-barabash, 2023-07-10:
That makes total sense Christopher. It was my thinking as well and I ignored it for some time. In retrospect, I admit that my testing wasn't (academically) rigorous but I played for weeks manually adding the mouse using bluetoothctl and never was able to get it to survive reboots/standby. Perhaps it was just coincidental that adding the  `systemctl enable --now bluetooth` step suddenly seemed to help. The mouse is now solid on two computers that had the problem. Perhaps somewhere within the steps is a nugget? Like the saying goes, "even a blind squirrel finds a nut sometimes" LOL.

`systemctl enable --now bluetooth` does seem to inform about some links/files that are created and run. Is it possible that certain files/links aren't completely formed when using the GUI whereas this manual step forces them to be recreated? 

I don't know ... this is well beyond my level of expertise.

cschramm, 2023-07-10:
I'd infer that something was not properly set up if the command created additional symlinks, yes. I would not expect Bluetooth to work at all then, though. Weird.

Anyway, people experience a couple of issues with BlueZ that can be fixed by removing state from `/var/lib/bluetooth` and purging the package is one radical way to do that. Fingers crossed that your issue is gone, but I cannot imagine what could have fixed it.

One source of trouble when it's related to reconnecting to paired devices is a shared encryption key e.g. when dual booting multiple systems with the same Bluetooth dongle. Other sources of course include broken driver and firmware. You might be able to get some help from the BlueZ project but Linux Mint is far behind upstream sources and debugging is usually only done on recent versions.

In case removing things from `/var/lib/bluetooth` works, make a backup and compare the states to possibly find the crucial change.

closed 2023-09-16 by github-actions[bot]