blueman-project/blueman #2093: Device doesn't show up

JHudd435, 2023-06-20:
blueman: 2.3.5-2
BlueZ: 5.66-1
Distribution: Arch Linux
Desktop environment: i3, XFCE, Cinnamon

When I run blueman-manager, it shows some devices, but not my JLab Go Earbuds. The earbuds say ready to pair, but do not show up in the blueman menu. There are several devices in the menu that start with "GVH" with random numbers and letters following.

JHudd435, 2023-06-20:
There are no errors when I run troubleshooting

cschramm, 2023-06-20:
Does the device show up in debug logging?

Please check if unhiding unnamed devices in the View menu helps here.

JHudd435, 2023-06-20:
I checked unnamed devices, and tried to connect to every device. Connection failed for each one. Here is the debug log:

```
blueman-applet 09.37.55 INFO     PluginManager:87 load_plugin: ['RecentConns', 'PowerManager', 'StandardItems', 'NetUsage', 'GameControllerWakelock', 'StatusNotifierItem', 'SerialManager', 'Networking', 'ConnectionNotifier', 'PPPSupport', 'DhcpClient', 'DiscvManager', 'Menu', 'NMPANSupport', 'DisconnectItems', 'AuthAgent', 'KillSwitch', 'ExitItem', 'StatusIcon', 'ShowConnected', 'TransferService', 'NMDUNSupport', 'AutoConnect', 'DBusService']
blueman-applet 09.37.55 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 09.37.55 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 09.37.55 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Address': 'CC:3D:82:DE:E8:B9', 'AddressType': 'public', 'Name': 't540p', 'Alias': 't540p', 'Class': 1048844, 'Powered': True, 'Discoverable': False, 'DiscoverableTimeout': 180, 'Pairable': False, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['0000110e-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001133-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001'], 'Modalias': 'usb:v1D6Bp0246d0542', 'Roles': ['central', 'peripheral']}
blueman-applet 09.37.55 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 09.37.55 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 09.37.55 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 09.37.55 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 09.37.55 WARNING  PluginManager:151 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 09.37.55 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 09.37.55 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem'>
blueman-applet 09.37.55 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 09.37.55 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 09.37.55 INFO     Networking:35 load_nap_settings: Loading NAP settings
blueman-applet 09.37.55 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 09.37.55 WARNING  PluginManager:151 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 09.37.55 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 09.37.55 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 09.37.55 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 09.37.55 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 09.37.55 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 09.37.55 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 09.37.55 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 09.37.55 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 09.37.55 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 09.37.55 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 09.37.55 INFO     KillSwitch:106 io_event  : killswitch registered 0
blueman-applet 09.37.55 INFO     KillSwitch:122 io_event  : State: True
blueman-applet 09.37.55 INFO     PowerManager:179 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 09.37.55 INFO     KillSwitch:106 io_event  : killswitch registered 1
blueman-applet 09.37.55 INFO     KillSwitch:122 io_event  : State: True
blueman-applet 09.37.55 INFO     PowerManager:179 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 09.37.55 INFO     KillSwitch:85 _on_connman_vanished: net.connman vanished
blueman-applet 09.37.55 INFO     TransferService:234 _on_dbus_name_appeared: org.bluez.obex :1.51
[... cut ...]
blueman-applet 09.38.35 INFO     Applet:100 on_device_removed: Device removed /org/bluez/hci0/dev_A4_C1_38_E8_FD_0A
blueman-applet 09.38.43 DEBUG    Manager:73 _on_object_removed: /org/bluez/hci0/dev_A4_C1_38_AE_64_91
blueman-applet 09.38.43 INFO     Applet:100 on_device_removed: Device removed /org/bluez/hci0/dev_A4_C1_38_AE_64_91
blueman-applet 09.39.01 DEBUG    Manager:73 _on_object_removed: /org/bluez/hci0/dev_A0_76_4E_30_8D_E2
blueman-applet 09.39.01 INFO     Applet:100 on_device_removed: Device removed /org/bluez/hci0/dev_A0_76_4E_30_8D_E2
```

JHudd435, 2023-08-20:
Further activity

cschramm, 2023-08-20:
Sorry, this drowned.

The only unnamed device I see in your log is `A0:76:4E:30:8D:E2`. I have no indication that it's the one you're looking for. Please check the device class in the info dialog from its right-click menu (also the icon should be a good indication; I'd expect something like `audio-headphones` for earbuds).

Assuming that device is not the one you are looking for, it does not seem to be visible ("in pairing mode"). Are you able to discover it from another device at the same time?

closed 2023-08-28 by github-actions[bot]