blueman-project/blueman #2095: [Feature] Create a flag to allow multiple instances of the tray icon.

aarondill, 2023-06-25:
<details>
<summary>Version Info</summary>

blueman:
```shell
> dpkg-query --showformat='Version: ${Version}' --show blueman
Version: 2.2.4-1
```
BlueZ:
```shell
> dpkg-query --showformat='Version: ${Version}' --show bluez
Version: 5.64-0ubuntu1
```
Distribution:
```
lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.2 LTS
Release:        22.04
Codename:       jammy
```
Desktop environment: No DE, [awesome window manager](https://github.com/awesomeWM/awesome/) v4.3
</details>

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I occasionally run multiple instances of my window manager (in seperate ttys) and would appreciate having full access to my system tray. This could be remedied by adding a flag (`--multiple-instances/-m` ?) to permit multiple instances of (at least) the tray icon, though, as `blueman-applet` is what's run, it would also likely be needed to permit multiple instances of the full applet.

This could be implemented by passing `Gio.ApplicationFlags.NON_UNIQUE` to `super().__init__` in the [tray](https://github.com/blueman-project/blueman/blob/main/blueman/main/Tray.py#L14C9-L14C25) as well as in the [Main applet](https://github.com/blueman-project/blueman/blob/main/blueman/main/Applet.py#L27).

aarondill, 2023-06-25:
This could look something like this (example for the tray) and would work because the flags are represented as a single integer which are bitwise OR'ed together (and `FLAGS_NONE` is `0`) 

```python
class BluemanTray(Gio.Application):
    def __init__(self) -> None:
        app_flags=Gio.ApplicationFlags.FLAGS_NONE
        if (ARGS_HAS_MULTIPLE_INSTANCES) # This needs implementation
            app_flags=app_flags | Gio.ApplicationFlags.NON_UNIQUE
        super().__init__(application_id="org.blueman.Tray", flags=app_flags)
        # ...
```

aarondill, 2023-06-25:
One additional note slightly unrelated to this PR, the Gio docs state that `FLAGS_NONE` is depreciated in favor of `DEFAULT_FLAGS` in version 2.74. [Gio.ApplicationFlags.FLAGS_NONE](https://lazka.github.io/pgi-docs/Gio-2.0/flags.html#Gio.ApplicationFlags.FLAGS_NONE)

cschramm, 2023-06-25:
If I get it right, you do have a single D-Bus session bus for multiple desktops, right? Why do you think we need multiple applet processes? One issue that immediately comes to my mind is that only one of them can register the default agent with BlueZ so that only one of the desktops would receive authorization requests and the like.

I absolutely can imagine multiple tray instances with a single applet instance, though.

Being curious: What's your experience with other "tray applications" and multiple window manager instances?

aarondill, 2023-06-25:
I don't particularly need (or want) multiple applet processes, however I figured it might take slightly more processing to launch several trays vs just brute force another applet.

Honestly, D-Bus is a mystery to me. I'm running `dbus-launch --exit-with-session -- startx /usr/bin/awesome` (automatically :smile:) to start each instance, so maybe i'm running separate D-Bus sessions? Idek :man_shrugging: \

My experience is exactly half and half funnily enough. `diodon`, `nm-applet`, and `blueman-applet` all only allow one instance, while `pasystray`, `udiskie`, and `ibus-daemon` all permit multiple instances.

aarondill, 2023-06-25:
A few other applications (albeit daemons) that allow multiple instances that I use are: `polkit-gnome-authentication-agent-1`, `picom`, `notification_daemon`, `xss-lock`.

While it's not super common that I run multiple instances, as I usually just use workspaces (like a sane person :smile:), it is useful for debugging my WM, and when I'm struggling with Awesome, It's helpful to have full access to the remainder of my system controls in my system bar.

infirit, 2023-06-26:
The current design is that the applet (re)starts the tray. It's not trivial to change this.

cschramm, 2023-06-27:
Yes, I absolutely would not expect it to work out of the box either, e.g. I don't think the communication between one applet instance and multiple tray instances would work as intended.

Anyway, you do seem to have one session bus per desktop. so it's obvious that we'd need one applet per desktop and applet <-> tray would still be 1:1. `NON_UNIQUE` should not be necessary either, as things are still unique within one session bus then. I'd actually expect it to pretty much work. The only problem you'll definitely see is that only one of the applets can be registered as the default agent to bluetoothd. The newer one wins, so that you'll not get requests on the older desktop as long as you do not restart blueman there.

I just tried and actually `dbus-launch blueman-applet --loglevel debug` works for me to start additional applet and tray instances (on the same desktop). As expected they do steal and become the default agent and I do see an error when they try to re-register a network server (`org.bluez.Error.AlreadyExists`).

What issue do you run into?

aarondill, 2023-06-27:
prefixing `blueman-applet` with `dbus-launch` *does* create a second instance, however, I currently run my *WM* under `dbus-launch`, and when running `blueman-applet --loglevel debug` there (without a further `dbus-launch`) it returns the same output as if run in the older wm, without starting a second instance.

<details>
<summary>Desktop 1: `> blueman-applet --loglevel debug`</summary>

```
blueman-applet 04.57.32 INFO     PluginManager:85 load_plugin: ['NMPANSupport', 'KillSwitch', 'StatusIcon', 'PPPSupport', 'Menu', 'ExitItem', 'TransferService', 'ShowConnected', 'DBusService', 'SerialManager', 'Battery', 'ConnectionNotifier', 'NMDUNSupport', 'DiscvManager', 'AppIndicator', 'DhcpClient', 'NetUsage', 'RecentConns', 'StandardItems', 'AuthAgent', 'GameControllerWakelock', 'DisconnectItems', 'PowerManager', 'AutoConnect', 'Networking']
blueman-applet 04.57.32 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 04.57.32 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 04.57.32 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Address': 'F0:A6:54:01:D6:A6', 'AddressType': 'public', 'Name': 'aaron-HP-Laptop-15-dy2xxx', 'Alias': 'aaron-HP-Laptop-15-dy2xxx', 'Class': 8126732, 'Powered': True, 'Discoverable': False, 'DiscoverableTimeout': 180, 'Pairable': True, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['0000110e-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '00001108-0000-1000-8000-00805f9b34fb', '00001133-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001'], 'Modalias': 'usb:v1D6Bp0246d0540', 'Roles': ['central', 'peripheral']}
blueman-applet 04.57.32 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 04.57.32 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 04.57.32 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 04.57.32 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 04.57.32 WARNING  PluginManager:147 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 04.57.32 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 04.57.32 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 04.57.32 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 04.57.32 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 04.57.32 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Battery.Battery'>
blueman-applet 04.57.32 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 04.57.32 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 04.57.32 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 04.57.32 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AppIndicator.AppIndicator'>
blueman-applet 04.57.32 WARNING  PluginManager:147 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 04.57.32 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 04.57.32 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 04.57.32 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 04.57.32 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 04.57.32 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 04.57.32 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 04.57.32 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 04.57.32 INFO     Networking:35 load_nap_settings: Loading NAP settings
blueman-applet 04.57.33 INFO     KillSwitch:106 io_event  : killswitch registered 1
blueman-applet 04.57.33 INFO     KillSwitch:122 io_event  : State: True
blueman-applet 04.57.33 INFO     PowerManager:178 update_power_state: off False | foff False | on True | current state True | new state True
[... cut ...]
blueman-applet 04.54.40 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.DBusService.DBusService object at 0x7fbe35c1a710>
blueman-applet 04.54.40 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <StatusIcon.StatusIcon object at 0x7fbe3708cdc0 (blueman+plugins+applet+StatusIcon+StatusIcon at 0x561e0f496da0)>
blueman-applet 04.54.40 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.Menu.Menu object at 0x7fbe35c1a6e0>

```
</details>

cschramm, 2023-06-27:
:thinking: I'm puzzled why it quits on the second desktop. Just tried and successfully started blueman on a separate desktop without any tricks (also tried 2.2.4 but I might have done something wrong there, so no guarantees). Are you sure that there's only a single blueman-applet process from the first one?

Does `Gio.ApplicationFlags.NON_UNIQUE` help to get it starting on the second desktop?

aarondill, 2023-06-27:
running this to replace both instances of FLAGS_NONE with NON_UNIQUE:

```shell
> cd /usr/lib/python3/dist-packages/blueman/
> rg 'Gio.ApplicationFlags.FLAGS_NONE' -l | xargs -n1 sudo sed -i 's/Gio.ApplicationFlags.FLAGS_NONE/Gio.ApplicationFlags.NON_UNIQUE/g' 
```

<details>
<summary>Desktop 1: `blueman-applet` (this is normal)</summary>

```
blueman-applet 13.53.09 WARNING  PluginManager:147 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 13.53.09 WARNING  PluginManager:147 __load_plugin: Not loading DhcpClient because its conflict has higher priority
```
</details>

<details>
<summary>Desktop 2: `blueman-applet` (Second Instance Started)</summary>

```
blueman-applet 13.51.43 WARNING  PluginManager:147 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 13.51.43 WARNING  PluginManager:147 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 13.51.43 ERROR    AgentManager:20 on_register_failed: /org/bluez/obex/agent/blueman org.bluez.obex.Error.AlreadyExists Agent already exists
```
</details>

Interestingly, if I *close* the second instance (ctrl+c), I get this error on the first:

```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/main/Tray.py", line 27, in _on_name_appeared
    indicator_name = applet.GetStatusIconImplementation()
  File "/usr/lib/python3/dist-packages/gi/overrides/Gio.py", line 349, in __call__
    result = self.dbus_proxy.call_sync(self.method_name, arg_variant,
gi.repository.GLib.GError: g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name :1.100 was not provided by any .service files (2)
```

aarondill, 2023-06-27:
After that error, the tray dies, but the applet seems to survive. manually running `blueman-tray` revives my tray icon and everything (seems) to return to working order.

aarondill, 2023-06-27:
I tested running two instances, and it seems to be able to (at least) do all the major actions (connect/disconnect) and both instances handle updating the icon and tray states fine.

aarondill, 2023-06-27:
Here is some debug output. Oddly enough, this time, I didn't get the tray exception...

Please note that the `----MESSAGE----` lines were added by me (using printf '%s\n' "MESSAGE" >>file).

<details>
<summary>Desktop 1: `blueman-applet  --loglevel debug` (this is normal)</summary>

```
blueman-applet --loglevel debug >>file1 2>&1 & PID=$! && echo $PIDblueman-applet 14.20.23 INFO     PluginManager:85 load_plugin: ['NMPANSupport', 'KillSwitch', 'StatusIcon', 'PPPSupport', 'Menu', 'ExitItem', 'TransferService', 'ShowConnected', 'DBusService', 'SerialManager', 'Battery', 'ConnectionNotifier', 'NMDUNSupport', 'DiscvManager', 'AppIndicator', 'DhcpClient', 'NetUsage', 'RecentConns', 'StandardItems', 'AuthAgent', 'GameControllerWakelock', 'DisconnectItems', 'PowerManager', 'AutoConnect', 'Networking']
blueman-applet 14.20.23 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 14.20.23 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 14.20.23 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Address': 'F0:A6:54:01:D6:A6', 'AddressType': 'public', 'Name': 'aaron-HP-Laptop-15-dy2xxx', 'Alias': 'aaron-HP-Laptop-15-dy2xxx', 'Class': 8126732, 'Powered': True, 'Discoverable': False, 'DiscoverableTimeout': 180, 'Pairable': False, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['0000110e-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '00001108-0000-1000-8000-00805f9b34fb', '00001133-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001'], 'Modalias': 'usb:v1D6Bp0246d0540', 'Roles': ['central', 'peripheral']}
blueman-applet 14.20.23 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 14.20.23 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 14.20.23 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 14.20.23 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 14.20.23 WARNING  PluginManager:147 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 14.20.23 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 14.20.23 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 14.20.23 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 14.20.23 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 14.20.23 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Battery.Battery'>
blueman-applet 14.20.23 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 14.20.23 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 14.20.23 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 14.20.23 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AppIndicator.AppIndicator'>
blueman-applet 14.20.23 WARNING  PluginManager:147 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 14.20.23 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 14.20.23 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 14.20.23 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 14.20.23 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 14.20.23 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 14.20.23 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 14.20.23 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 14.20.23 INFO     Networking:35 load_nap_settings: Loading NAP settings
blueman-applet 14.20.23 INFO     KillSwitch:106 io_event  : killswitch registered 1
blueman-applet 14.20.23 INFO     KillSwitch:122 io_event  : State: True
blueman-applet 14.20.23 INFO     PowerManager:178 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 14.20.23 INFO     KillSwitch:85 _on_connman_vanished: net.connman vanished
[... cut ...]
blueman-applet 14.21.20 INFO     Networking:71 set_nap   : set nap False
blueman-applet 14.21.20 ERROR    AgentManager:20 on_register_failed: /org/bluez/obex/agent/blueman org.bluez.obex.Error.AlreadyExists Agent already exists
blueman-applet 14.21.20 INFO     ShowConnected:50 enumerate_connections: Found 0 existing connections

(blueman-tray:21827): Gdk-CRITICAL **: 14:21:21.272: gdk_window_thaw_toplevel_updates: assertion 'window->update_and_descendants_freeze_count > 0' failed
---KILLING FIRST INSTANCE---
--KILLING-SECOND-INSTANCE--

```
</details>

cschramm, 2023-06-28:
Seems like the same session bus then. Without `NON_UNIQUE` a second instance fails to register the name and with non-unique instances the tray process is confused. :smile:

Within a single session bus the way to go would rather be a single applet with multiple tray instances, i.e. setting `NON_UNIQUE` for the tray application only and starting it for each desktop. A quick test with non-unique tray instances just worked fine on my end.

However, from https://github.com/blueman-project/blueman/issues/2095#issuecomment-1606000076 and my experiments with that, I'd clearly expect one session bus per desktop. :thinking:

aarondill, 2023-06-28:
I really couldn't explain to you my DBus setup. Like I said, Dbus is black magic to me, and I'm not even sure why I'm running dbus-launch. I saw it in the /xsession.d/* files, so I copied it into my start script.

cschramm, 2023-06-28:
Alright, so, assuming that the target is multi-desktop, single-session-bus, I think the way to go is setting `NON_UNIQUE` for the tray application (but not the applet) so that GLib does not get in your way when you start blueman-tray on each desktop (with the respective `$DISPLAY`). I guess it should just work, but there might be some issues, of course.

I agree that `NON_UNIQUE` should be behind a special option as in standard cases GLib does a good job for us to avoid duplicate icons. Probably there's exactly one user for the option, but it should not hurt, so why not. :smile:

infirit, 2023-06-28:
My thoughts go to using the `XDG_CURRENT_DESKTOP` environment variable to spawn new tray applets instead of `NON_UNIQUE`. Any well behaved desktop setup should set this. Not sure how well behaved these tiling ones are though 🤔

cschramm, 2023-06-28:
awesome does not seem to set that and even if it would I'd just expect the same value for all instances. Maybe I don't get the idea yet, though. :sweat_smile:

What should differ is `$DISPLAY`. Ideally tray would support at most one instance per `$DISPLAY`.

infirit, 2023-06-28:
Ooh, I didn't pickup that it's the same desktop 😅. Yeah, then DISPLAY may be a good alternative.