blueman-project/blueman #2096: Blueman spams the application toobar spam

stevef9432203, 2023-06-27:
blueman: 2.3.5
BlueZ:
Distribution: fedora 38
Desktop environment: mate

Using A bluetooth mouse the application bar gets  filled with mx mouse  application popups.  We need a way to autoclose  the notice window

cschramm, 2023-06-27:
What are "mx mouse application popups"?

Regarding desktop notifications, blueman expects timeouts to be supported and mate-notification-daemon does have support.

stevef9432203, 2023-06-27:
Like  Thus
[image]

closed 2023-06-27 by stevef9432203

cschramm, 2023-06-27:
That looks like our fallback. Is your notification daemon fine?

`dbus-send --session --print-reply --dest=org.freedesktop.Notifications /org/freedesktop/Notifications org.freedesktop.Notifications.GetServerInformation`