blueman-project/blueman #2097: [Feature] Hide blocked devices

anasoid, 2023-07-04:
blueman: 2.3.5
BlueZ: 
Distribution: linux mint 21
Desktop environment: cinnamon 5.6

I have many neighbors with bluetooth device (phone, watch, TV) So i block all there device on blueman manager but i have a big list on it, there is no way to hide blocked devices?

infirit, 2023-07-04:
There is currently no way.

I do wonder why you decided to block them as this is likely the reason they remain in the list.

anasoid, 2023-07-05:
I connect regularity Bluetooth device, so if i doesn't block them some time by error if I click on the wrong line i will be connected to an unknown device

infirit, 2023-07-05:
How about allowing you to hide any device without blocking it.

anasoid, 2023-07-05:
thanks , yes it will be better

DHager, 2025-12-09:
Yes please, from my apartment there are always neighbors' devices which I will *never* want to connect to.  (Currently 6 named, 17 unnamed.) 

So all those wrong (perhaps even dangerous) choices will clutter things up when I'm looking fora new device of my own, and if I "block" them for safety then they just show up even more.