blueman-project/blueman #2099: Change the tray icon for bluetooth icons from default in epapriusdark in Mint to ones for Numix

adrummer354, 2023-07-09:
blueman: 2.3.5
BlueZ: bluetoothctl: 5.64
Distribution: Linux Mint 21.1 
Desktop environment: Cinnamon 

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2023-07-09:
I don't think I understand the request or question. I assume the Papirus icon theme is your system's default and you're seeing its blueman icons but you want to get the ones from Numix. The solution is ... to use Numix.

adrummer354, 2023-07-09:
I think I maybe didn't explain it very clearly. I like all of the icons from epapirus-dark theme except for the status tray icon that shows up for the blueman-applet icons which doesn't change when I use epaprius-dark. I want to keep my icon theme as epapirus dark for most of my icons and I don't want to use and of the icons from Numix except I want to see Numix's style of icons for the blueman-applet in my system tray. I think that the blueman-applet/blueman-manager overrides my set epaprius-dark's icon theme's preferences and that blueman applet maybe uses gnome's icons which are very dull and don't fit my desired icons which are the ones for numix/epaprius. I could send you a link to see what I mean. Have a look at this:       https://imgur.com/a/XsMBw4Z

cschramm, 2023-07-10:
The top-left icon in that image actually does not look like Papirus. I see blueman icons in Papirus, Papirus-Light and ePapirus. Unless the dark themes have a fallback to those, it seems like they do not have blueman icons at all (or even Bluetooth in general?!). I'd call that a bug.

As a workaround you can provide any icons that you prefer as blueman-tray, blueman-active and blueman-disabled. The mentioned themes typically use bluetooth-active for both tray and active as well as bluetooth-disabled.

cschramm, 2023-07-10:
With Adwaita active I can e.g. provide an icon at `~/.icons/Adwaita/scalable/status/blueman-active.svg`. I guess with ePapirus-Dark `~/.icons/ePapirus-Dark/24x24/status/blueman-active.svg` should work.

In Linux Mint you might be using symbolic icons (check `dconf read /org/blueman/general/symbolic-status-icons`). Either `~/.icons/ePapirus-Dark/symbolic/status/blueman-active-symbolic.svg` or `~/.icons/Papirus/symbolic/status/blueman-active-symbolic.svg` could work then. It seems like Papirus does not have symbolic icon names for blueman yet, so it would be kind of a configuration error to use Papirus with `symbolic-status-icons` enabled (default is disabled).

adrummer354, 2023-07-10:
I'm not sure then what I would do. I tried to find that path you suggested for checking that dconf in /org/blueman/general/symbolic-status-icons but that doesn't look like that's a valid file path on my system. Files for icons are mostly showing in: /usr/share/icons/

I went back into the index.theme file in the epaprius-dark icon theme folder in that path and it shows that it inherts this way:

Inherits=elementary,hicolor

adrummer354, 2023-07-11:
I tried coping that file and other blueman files to the 24x24/status/ folder with elevated privileges directly from the numix icons folder and it gives me a broken link error and it doesn't look like the icon appears:

https://imgur.com/a/BM4sTbc

It's looking like those icons aren't the original file location I'm not sure where they would be.

cschramm, 2023-07-12:
Not sure I can follow, but blueman icons are often just links to bluetooth icons. If you copy in those links, the files they point to have to be present as well, of course.

Judging from your images, you seem to be using symbolic icons. With symbolic status icons enabled in blueman and ePapirus-Dark in my window manager, the following works fine for me to get the Numix icons:

```
mkdir -p ~/.icons/ePapirus-Dark/symbolic/status
wget https://raw.githubusercontent.com/numixproject/numix-icon-theme/master/Numix/scalable/status/bluetooth-active-symbolic.svg -O ~/.icons/ePapirus-Dark/symbolic/status/blueman-tray-symbolic.svg
wget https://raw.githubusercontent.com/numixproject/numix-icon-theme/master/Numix/scalable/status/bluetooth-paired-symbolic.svg -O ~/.icons/ePapirus-Dark/symbolic/status/blueman-active-symbolic.svg
wget https://raw.githubusercontent.com/numixproject/numix-icon-theme/master/Numix/scalable/status/bluetooth-disabled-symbolic.svg -O ~/.icons/ePapirus-Dark/symbolic/status/blueman-disabled-symbolic.svg
```

adrummer354, 2023-07-12:
I'm going to go ahead and close this out since I'm happy with what happened when I copied my desired icons to the /hicolor/scalable/status folder and then restarted cinnamon. Its looking like I wanted it to now! Much better. Thanks for helping me!

closed 2023-07-12 by adrummer354