blueman-project/blueman #2101: blueman-manager crashed with AttributeError in make_device_icon(): 'NoneType' object has no attribute 'load_surface'

tkrah, 2023-08-02:
blueman: 2.3.5+mint1+vera
BlueZ: 5.64-0ubuntu1
Distribution: Mint 21.1
Desktop environment: Cinnamon

* [ X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Using blueman-manager it is crashing regularly with apport telling me this as a title:

`blueman-manager crashed with AttributeError in make_device_icon(): 'NoneType' object has no attribute 'load_surface'`

I have no idea what's wrong here, anyone got an idea how to debug / fix that problem?

cschramm, 2023-08-03:
Please provide the traceback (or at least the location of the error).

From looking at the code I think that could be possible if you lack a device icon. Can you start blueman-manager again right away? I'd expect it to crash again as the same device should still be in the list.

tkrah, 2023-08-03:
Does this help:

```
PythonArgs: ['/usr/bin/blueman-manager']                              
Traceback:                                                            
Traceback (most recent call last):                                   
   File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 648, in _set_cell_data
      surface = self.make_device_icon(row["icon_info"], row["paired"], row["trusted"], row["blocked"])
    File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 265, in make_device_icon
      target = icon_info.load_surface(window)                          
  AttributeError: 'NoneType' object has no attribute 'load_surface'
```

I can restart it right away, it does not always crash with this, but it happens here and there, sometimes 2 times a day, sometimes its crashing more than 10 times a day.

cschramm, 2023-08-03:
Perfect, thanks.

So, `row["icon_info"]` seems to be `None`. Assuming that it gets explicitly set to that, let's try to catch it with an assertion. Can you please apply the following patch to `/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py` and see where that gets us:

```diff
--- /usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py
+++ /usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py
@@ -360,6 +360,7 @@ class ManagerDeviceList(DeviceList):
             description = get_major_class(device['Class'])
 
         icon_info = self.get_icon_info(device["Icon"], 48, False)
+        assert icon_info is not None, f"Failed to load icon {device['Icon']}"
         caption = self.make_caption(device['Alias'], description, device['Address'])
 
         self.set(tree_iter, caption=caption, icon_info=icon_info, alias=device['Alias'], objpush=has_objpush)

```

tkrah, 2023-08-03:
```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py", line 118, in __on_manager_signal
    self.device_add_event(dev)
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 316, in device_add_event
    self.add_device(device)
  File "/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py", line 238, in add_device
    self.row_setup_event(tree_iter, device)
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 363, in row_setup_event
    assert icon_info is not None, f"Failed to load icon {device['Icon']}"
AssertionError: Failed to load icon audio-card
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 127, in on_icon_theme_changed
    self.row_setup_event(row.iter, device)
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 363, in row_setup_event
    assert icon_info is not None, f"Failed to load icon {device['Icon']}"
AssertionError: Failed to load icon audio-card
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 649, in _set_cell_data
    surface = self.make_device_icon(row["icon_info"], row["paired"], row["trusted"], row["blocked"])
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 265, in make_device_icon
    target = icon_info.load_surface(window)
AttributeError: 'NoneType' object has no attribute 'load_surface'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 649, in _set_cell_data
    surface = self.make_device_icon(row["icon_info"], row["paired"], row["trusted"], row["blocked"])
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 265, in make_device_icon
    target = icon_info.load_surface(window)
AttributeError: 'NoneType' object has no attribute 'load_surface'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 649, in _set_cell_data
    surface = self.make_device_icon(row["icon_info"], row["paired"], row["trusted"], row["blocked"])
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 265, in make_device_icon
    target = icon_info.load_surface(window)
AttributeError: 'NoneType' object has no attribute 'load_surface'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 649, in _set_cell_data
    surface = self.make_device_icon(row["icon_info"], row["paired"], row["trusted"], row["blocked"])
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 265, in make_device_icon
    target = icon_info.load_surface(window)
AttributeError: 'NoneType' object has no attribute 'load_surface'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 649, in _set_cell_data
    surface = self.make_device_icon(row["icon_info"], row["paired"], row["trusted"], row["blocked"])
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 265, in make_device_icon
    target = icon_info.load_surface(window)
AttributeError: 'NoneType' object has no attribute 'load_surface'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 649, in _set_cell_data
    surface = self.make_device_icon(row["icon_info"], row["paired"], row["trusted"], row["blocked"])
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 265, in make_device_icon
    target = icon_info.load_surface(window)
AttributeError: 'NoneType' object has no attribute 'load_surface'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 649, in _set_cell_data
    surface = self.make_device_icon(row["icon_info"], row["paired"], row["trusted"], row["blocked"])
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 265, in make_device_icon
    target = icon_info.load_surface(window)
AttributeError: 'NoneType' object has no attribute 'load_surface'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 649, in _set_cell_data
    surface = self.make_device_icon(row["icon_info"], row["paired"], row["trusted"], row["blocked"])
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 265, in make_device_icon
    target = icon_info.load_surface(window)
AttributeError: 'NoneType' object has no attribute 'load_surface'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 649, in _set_cell_data
    surface = self.make_device_icon(row["icon_info"], row["paired"], row["trusted"], row["blocked"])
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 265, in make_device_icon
    target = icon_info.load_surface(window)
[... 138 lines cut ...]
AttributeError: 'NoneType' object has no attribute 'load_surface'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 649, in _set_cell_data
[... cut ...]
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 649, in _set_cell_data
    surface = self.make_device_icon(row["icon_info"], row["paired"], row["trusted"], row["blocked"])
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 265, in make_device_icon
    target = icon_info.load_surface(window)
AttributeError: 'NoneType' object has no attribute 'load_surface'

```

Seems to be:

`AssertionError: Failed to load icon audio-card`

infirit, 2023-08-03:
It almost looks like your icon theme is missing that icon. Can you try using a different icon theme, preferably something like adwaita as it should have it.

Generally, we should probably never allow icon_info to be None and instead use the image-missing or blueman icon. The lookup allows for a list of names so that's fairly easy to handle.

tkrah, 2023-08-03:
Used adwaita, no errors anymore (although it's not my cup of tea so I'll change it back) - so it would be nice if you can implement that fallback to a known image.

cschramm, 2023-08-03:
I am confused on multiple levels...

Shouldn't Adwaita icons always be available, no matter if the active icon theme provides them as well or not? I thus would not expect switching to Adwaita to fix anything.

The error is triggered by some device for which BlueZ provides the icon name `audio-card` and the system has trouble loading that icon. I would expect an immediate crash after restarting blueman-manager as the same device should still be present with the same icon name. If it does not crash, the problem does not seem to happen every time that icon is requested.

What theme do you use?

tkrah, 2023-08-03:
I used to use 'Mint-Y-Legacy-Dark-Aqua' - the problem now is, I can't reproduce it anymore - after changing to Adwaita I can change back now to the mint theme and it does not crash with the missing icon info. Before changing it always had that bunch of errors above, strange ... still trying to reproduce this ... I'll tell back if I am able or unable to.

tkrah, 2023-08-03:
Tried my best the last 40 minutes (changing to different themes back and forth, reloading the WM, tried login / logout ... nothing) - since changing themes I am unable to crash it now again with that trace / errors from above, which is good for me in the end but I really can't tell you now anymore how to reproduce that, baffling ... sorry for the noise.

cschramm, 2023-08-03:
You're welcome. I actually expected that in my confusion, so at least I'm confident now that it is not expected behavior. :sweat_smile:

closed 2023-08-03 by cschramm

infirit, 2023-08-03:
I have some changes for the list so I'll add one more for this case as imo we should handle this more graceful.

cschramm, 2023-08-03:
@infirit: I'm not convinced that we should change anything there. To my understanding BlueZ always provides Adwaita icon names and GTK+ guarantees them to be available. Anything else is a bug, misconfiguration or - as in this case - some issue with the system's state. Of course we can add a fallback, but it should never come into action.