blueman-project/blueman #2107: Bluetooth device will not show up in Linux mint 21.2 cinnamon

jouletrix, 2023-08-06:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

jouletrix, 2023-08-06:
I was able to Connect my Device before but later i was not able to so i reinstalled the bluetooth now the device will not show

cschramm, 2023-08-15:
You are searching for devices and not discovering the one you are looking for although it is pairable and you are able to discover it from other systems and you are able to discover other devices from the same system at the same time? What kind of device is it? Did you try bluetoothctl?

closed 2023-08-30 by github-actions[bot]