blueman-project/blueman #2113: Can't turn off tray icon

angrybacon, 2023-08-10:
blueman: 2.3.5-2
BlueZ: 5.68-1
Distribution: Arch
Desktop environment: Qtile 0.22.1

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Was the feature reverted? https://github.com/blueman-project/blueman/pull/1496

[image]

cschramm, 2023-08-15:
It's unreleased.

angrybacon, 2023-08-15:
Any specific reason why?

I know there were some concerns on difficulty to re-enable the tray icon once disabled but I think re-enabling it the same way you disabled it in the first place is intuitive enough

cschramm, 2023-08-15:
There just has not been any feature release. It's not thoroughly tested (you're welcome to test!), but we're not really concerned. 😄

angrybacon, 2023-08-15:
Thanks for letting me know

Apologies if the error is obvious but I'm getting this when trying the build

```
$ blueman-manager
Traceback (most recent call last):
  File "/usr/local/bin/blueman-manager", line 15, in <module>
    from blueman.main.Manager import Blueman
ModuleNotFoundError: No module named 'blueman'
```

The steps I followed:

```sh
yay -R blueman
./autogen.sh
make
sudo make install
```

Thanks in advance

angrybacon, 2023-08-16:
@infirit Ah yes of course thank you :pray: 

So far so good, unsure how much testing from different users you need and for how long (it is pretty old code at this point) but it's fine for me with the Git package.

closed 2023-10-24 by github-actions[bot]