blueman-project/blueman #2114: br-connection-unknown issue keeps reappearing

RShah126, 2023-08-13:
blueman: 2.3.5
BlueZ: 5.64
Distribution: Mint
Desktop environment: X-Cinnamon

* I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I have tried repairing and reconnecting my bluetooth device. I'm running on a dual-boot system, but to no avail it's not connecting and pairing and its displaying "br-connection-unknown" I've even restarted bluetooth.    
[image]

RShah126, 2023-08-13:
[](url)
[image]
Also get pairing failed problem along with it, despite my headphones being recognized.

RShah126, 2023-08-13:
New problem, it's now also pairing and connecting but the sound is not going to the headphones

cschramm, 2023-08-15:
Check the bluetooth.service logs. They should have some information on the error. Possibly also dmesg.

RShah126, 2023-08-15:
Hello, how do I check for bluetooth.service logs

cschramm, 2023-08-15:
Something like `journalctl -u bluetooth.service`.

RShah126, 2023-08-15:
[logs.txt](https://github.com/blueman-project/blueman/files/12341872/logs.txt)

RShah126, 2023-08-15:
Jul 23 02:37:26 skywalker systemd[1]: Starting Bluetooth service...
Jul 23 02:37:26 skywalker bluetoothd[818]: Bluetooth daemon 5.64
Jul 23 02:37:27 skywalker systemd[1]: Started Bluetooth service.
Jul 23 02:37:27 skywalker bluetoothd[818]: Starting SDP server
Jul 23 02:37:27 skywalker bluetoothd[818]: Bluetooth management interface 1.21 initialized
Jul 23 02:37:44 skywalker bluetoothd[818]: Endpoint registered: sender=:1.48 path=/MediaEndpoint/A2DPSource/ldac_hq
Jul 23 02:37:44 skywalker bluetoothd[818]: Endpoint registered: sender=:1.48 path=/MediaEndpoint/A2DPSource/ldac_sq
Jul 23 02:37:44 skywalker bluetoothd[818]: Endpoint registered: sender=:1.48 path=/MediaEndpoint/A2DPSource/ldac_mq
Jul 23 02:37:44 skywalker bluetoothd[818]: Endpoint registered: sender=:1.48 path=/MediaEndpoint/A2DPSink/aptx_hd
Jul 23 02:37:44 skywalker bluetoothd[818]: Endpoint registered: sender=:1.48 path=/MediaEndpoint/A2DPSource/aptx_hd
Jul 23 02:37:44 skywalker bluetoothd[818]: Endpoint registered: sender=:1.48 path=/MediaEndpoint/A2DPSink/aptx
Jul 23 02:37:44 skywalker bluetoothd[818]: Endpoint registered: sender=:1.48 path=/MediaEndpoint/A2DPSource/aptx
Jul 23 02:37:44 skywalker bluetoothd[818]: Endpoint registered: sender=:1.48 path=/MediaEndpoint/A2DPSink/sbc
Jul 23 02:37:44 skywalker bluetoothd[818]: Endpoint registered: sender=:1.48 path=/MediaEndpoint/A2DPSource/sbc
Jul 23 02:37:44 skywalker bluetoothd[818]: Endpoint registered: sender=:1.48 path=/MediaEndpoint/A2DPSink/sbc_xq_453
Jul 23 02:37:44 skywalker bluetoothd[818]: Endpoint registered: sender=:1.48 path=/MediaEndpoint/A2DPSource/sbc_xq_453
Jul 23 02:37:44 skywalker bluetoothd[818]: Endpoint registered: sender=:1.48 path=/MediaEndpoint/A2DPSink/sbc_xq_512
Jul 23 02:37:44 skywalker bluetoothd[818]: Endpoint registered: sender=:1.48 path=/MediaEndpoint/A2DPSource/sbc_xq_512
Jul 23 02:37:44 skywalker bluetoothd[818]: Endpoint registered: sender=:1.48 path=/MediaEndpoint/A2DPSink/sbc_xq_552
Jul 23 02:37:44 skywalker bluetoothd[818]: Endpoint registered: sender=:1.48 path=/MediaEndpoint/A2DPSource/sbc_xq_552
Jul 23 02:37:51 skywalker bluetoothd[818]: Endpoint unregistered: sender=:1.48 path=/MediaEndpoint/A2DPSource/ldac_hq
Jul 23 02:37:51 skywalker bluetoothd[818]: Endpoint unregistered: sender=:1.48 path=/MediaEndpoint/A2DPSource/ldac_sq
Jul 23 02:37:51 skywalker bluetoothd[818]: Endpoint unregistered: sender=:1.48 path=/MediaEndpoint/A2DPSource/ldac_mq
Jul 23 02:37:51 skywalker bluetoothd[818]: Endpoint unregistered: sender=:1.48 path=/MediaEndpoint/A2DPSink/aptx_hd
Jul 23 02:37:51 skywalker bluetoothd[818]: Endpoint unregistered: sender=:1.48 path=/MediaEndpoint/A2DPSource/aptx_hd
Jul 23 02:37:51 skywalker bluetoothd[818]: Endpoint unregistered: sender=:1.48 path=/MediaEndpoint/A2DPSink/aptx
Jul 23 02:37:51 skywalker bluetoothd[818]: Endpoint unregistered: sender=:1.48 path=/MediaEndpoint/A2DPSource/aptx
Jul 23 02:37:51 skywalker bluetoothd[818]: Endpoint unregistered: sender=:1.48 path=/MediaEndpoint/A2DPSink/sbc
Jul 23 02:37:51 skywalker bluetoothd[818]: Endpoint unregistered: sender=:1.48 path=/MediaEndpoint/A2DPSource/sbc
Jul 23 02:37:51 skywalker bluetoothd[818]: Endpoint unregistered: sender=:1.48 path=/MediaEndpoint/A2DPSink/sbc_xq_453
Jul 23 02:37:51 skywalker bluetoothd[818]: Endpoint unregistered: sender=:1.48 path=/MediaEndpoint/A2DPSource/sbc_xq_453
Jul 23 02:37:51 skywalker bluetoothd[818]: Endpoint unregistered: sender=:1.48 path=/MediaEndpoint/A2DPSink/sbc_xq_512
Jul 23 02:37:51 skywalker bluetoothd[818]: Endpoint unregistered: sender=:1.48 path=/MediaEndpoint/A2DPSource/sbc_xq_512
Jul 23 02:37:51 skywalker bluetoothd[818]: Endpoint unregistered: sender=:1.48 path=/MediaEndpoint/A2DPSink/sbc_xq_552
Jul 23 02:37:51 skywalker bluetoothd[818]: Endpoint unregistered: sender=:1.48 path=/MediaEndpoint/A2DPSource/sbc_xq_552
Jul 23 02:37:52 skywalker bluetoothd[818]: Endpoint registered: sender=:1.67 path=/MediaEndpoint/A2DPSource/ldac_hq
Jul 23 02:37:52 skywalker bluetoothd[818]: Endpoint registered: sender=:1.67 path=/MediaEndpoint/A2DPSource/ldac_sq
Jul 23 02:37:52 skywalker bluetoothd[818]: Endpoint registered: sender=:1.67 path=/MediaEndpoint/A2DPSource/ldac_mq
Jul 23 02:37:52 skywalker bluetoothd[818]: Endpoint registered: sender=:1.67 path=/MediaEndpoint/A2DPSink/aptx_hd
Jul 23 02:37:52 skywalker bluetoothd[818]: Endpoint registered: sender=:1.67 path=/MediaEndpoint/A2DPSource/aptx_hd
Jul 23 02:37:52 skywalker bluetoothd[818]: Endpoint registered: sender=:1.67 path=/MediaEndpoint/A2DPSink/aptx
Jul 23 02:37:52 skywalker bluetoothd[818]: Endpoint registered: sender=:1.67 path=/MediaEndpoint/A2DPSource/aptx
Jul 23 02:37:52 skywalker bluetoothd[818]: Endpoint registered: sender=:1.67 path=/MediaEndpoint/A2DPSink/sbc
Jul 23 02:37:52 skywalker bluetoothd[818]: Endpoint registered: sender=:1.67 path=/MediaEndpoint/A2DPSource/sbc

cschramm, 2023-08-20:
Are you sure that you tried to pair in that log? Then actually no error shows up when you get the br-connection-unknown. Unexpected but not impossible. I'd try debug logging then. https://wiki.ubuntu.com/DebuggingBluetooth#Generating_log_files_with_debugging_information has instructions for your upstream distribution.

RShah126, 2023-08-20:
Hello, for some reason it's now able to pair

closed 2023-10-28 by github-actions[bot]