blueman-project/blueman #2115: German translation could use some improvement

geos-github, 2023-08-13:
blueman: 2.3.5
BlueZ: 5.68
Distribution: Manjaro
Desktop environment: XFCE

[ X ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

The German translation of Blueman (https://github.com/blueman-project/blueman/blob/main/po/de.po) could use some improvements:

It translates:

msgid "_Make Discoverable"
msgstr "_Erkennbar machen"

However, it is very uncommon to use the word "erkennbar" (which means something like 'recognizable') in the context of Bluetooth. The more natural term is "Sichtbar machen".

And indeed Blueman translates:

msgid "Amount of time in seconds discoverable mode will last"
msgstr "Zeitspanne (in Sekunden), die der Adapter sichtbar sein wird"

as well as:

msgid "Discoverable… %ss"
msgstr "Sichtbar … %ss"

So, also for consistency, please use the term "sichtbar" throughout the software, do not use "erkennbar" (it's used only once as far as I could see).

Another issue:

msgid "Discoverable timeout"
msgstr "Feststellbarer Timeout"

This is an incorrect translation. In English (which is somewhat more flexible, one might argue :-)), the term indicates "the timeout of the discoverable state " or "the time until the discoverable state times out", in short simply "discoverable timeout".
Please note that "discoverable" here is not an adjective to "timeout", although it easily appears so at first glance.
This is exactly what the translator mistook it for and therefore produced a translation which does not make sense.  It reads as "timeout which can be recognized/established". Which is nonsense of course.
A correct (but maybe linguistically somewhat poor) translation would be "Sichtbarkeits-Timeout".

If you look at the French translation, for example, it reads:

msgid "Discoverable timeout"
msgstr "Temps limite de visibilité"

Note that they use the term "visible", and not something like "découvrable".

The German translation should also do it that way.

When you look at the code in blueman/plugins/applet/DiscvManager.py there  is even a definition:
"Amount of time in seconds discoverable mode will last".

Nice.

And nicely translated into German:

msgid "Amount of time in seconds discoverable mode will last"
msgstr "Zeitspanne (in Sekunden), die der Adapter sichtbar sein wird"

So, why not call "Discoverable timeout" simply "Sichtbarkeitsdauer" (literally "visibility duration")?
This is in my option the most appropriate and natural term in German to use in that context.

It should be trivial to make those two changes to blueman/blob/main/po/de.po and it would surely improve quality.

geos-github, 2023-08-18:
I signed up and Weblate and made the changes there. Is that all it takes?

infirit, 2023-08-19:
Pretty much. At some point weblate will push the translation to GitHub. We merge it before releases or when enough has accumulated in main.

infirit, 2023-08-23:
Your changes got pushed to the translation PR's.

closed 2023-08-23 by infirit