blueman-project/blueman #2116: Blueman-manager: Newly discovered devices' names do not get displayed; error on retrieving it

geos-github, 2023-08-13:
blueman: 2.3.5
BlueZ: 5.68
Distribution: Manjaro
Desktop environment: XFCE

[ X ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

(please note that this is most likely an update to #944, which was closed due to inactivity without being resolved. The issue still remains; therefore, as I cannot reopen it, I'm creating this new issue with hopefully more precise information that what the original poster provided a few years ago)

Setup: PC with up-to-date Manjaro XFCE, Blueman-Manager 2.3.5. Bluetooth USB adapter, lsusb says "Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)".
When I perform a search in Blueman-Manager the Bluetooth Dongle is activated (green LED is on). When I put an old Android phone with BT turned on and set to temporarily discoverable near the BT adapter, it is being found by Blueman manager and correctly displayed as a device type smartphone, but instead of its name its Bluetooth MAC address is displayed. This happens with other Bluetooth devices, too (e.g. a Mac, which is correctly recognized as desktop computer), so it looks like it is not specific to the device found.
Interestingly I get the following message in dmesg:

Bluetooth: hci0: unexpected cc 0x041a length: 7 > 1

This happens for every search which hits an "unknown" device.

Interestingly, for device (also an Android phone) which has been paired before, Blueman-manager search does display the full device name and the above message does not show up in dmesg, even when the pairing has been removed in the meantime on the phone. That device (including its name) remains stored in /var/lib/bluetooth/(adapter's MAC)/cache/, so I guess Blueman-manager simply reads out the name from that cache insteady of trying to query it which somehow fails.
After deleting the entry from the cache a Blueman-manager search on that previously paired phone again displays it with its MAC only, and the error message is added to dmesg.

The key question is thus: what does the above error message mean, i.e. what is going wrong when trying to retrieve a newly discovered device's name?

cschramm, 2023-08-14:
blueman is just a frontend to BlueZ and shows what BlueZ' API provides. We do not read anything from /var/lib/bluetooth.

Regarding the error message there are few hits on the Internets, mostly connected to dual booting Windows with fastboot enabled.

If I get you right, the name only gets resolved when pairing the device, correct? We know that for some keyboards, but it sounds to happen with all kinds of devices for you? Sounds like some firmware issue to me.

cschramm, 2023-08-16:
0A12:0001 are often counterfeit chips causing all kinds of issues.

Linux tries to match them as good as possible, but that's an endless fight of course. See e.g. https://patchwork.kernel.org/project/bluetooth/patch/0bba3f22-a232-3c07-1b05-73e6d38dab8a@gmail.com/.

geos-github, 2023-08-16:
Thanks for the link.
This is what the kernel (6.1.44) says on startup:

Bluetooth: hci0: unexpected event for opcode 0x0000
Bluetooth: hci0: CSR: Setting up dongle with HCI ver=3 rev=07a6; LMP ver=3 subver= 07a6; manufacturer=10

Top me this looks like it is not identified as a fake version, but as a legit BT 2.0 dongle, right (in line with the comments in the source could you provided a link to).
I opened the dongle, and the Chip has a CSR logo and the following inscription printed on it: 41B14 3A05U 549AE

Why am I getting the  "unexpected event for opcode 0x0000" message as well as later on the "unexpected cc 0x041a length: 7 > 1" messages? (the latter does not occur with some older Linux distributions I tried, but still the scan only yields the MAC, not the name, for unknown devices)

cschramm, 2023-08-20:
Looks absolutely legit from that data, yes.

Might be worth filing a kernel bug, but you should definitely try a more recent kernel first. Does Manjaro offer something there (6.4 at least)?

As an alternative you could contact the BlueZ project to further look into it, see http://www.bluez.org/contact/ for mailing list, Slack and IRC, or via [GitHub issues](https://github.com/bluez/bluez/issues).

geos-github, 2023-08-20:
Manjaro has a niche feature for switching kernels. I tried it with 6.4.9 as well as with 6.5.0rc5. Both yield the same result as 6.1.44.

The BT dongle I'm using is the one found here, BTW:
https://fccid.io/SHP-BED009/Label/Label-Location-and-Sample-572556

cschramm, 2023-08-20:
Perfect, I recommend filing a kernel bug with the findings then.

geos-github, 2023-08-21:
filed https://bugzilla.kernel.org/show_bug.cgi?id=217812