blueman-project/blueman #2118: cannot established BT connection over GUI

antondrroid, 2023-08-17:
blueman:yes
BlueZ:
Distribution:debian 12
Desktop environment:xfce

1.I had tried to connect my devices over BTmanager, it has device connected, then (after 2-5 seconds) disconnected, and after few attempts it  refused even to pair and to trust, (all 3 devices - keyboard, mouse and headphones) I tried about 30 min and make about 30 attempts.

while by CLI over blutothctl cmd it has all 3 paired, trusted & connected, sometimes from the first attempt, sometimes 2nd or 3rd.

2.i had wrote the data of all the connected devices into file /etc/bluetooth/hcid.conf, (4 items -2 mouses, 1 kbrd, 1 HdPhones), then i commented 2 of them (1 headphones and 1 mouse) and rebooted.
I had expected, that after boot will be work 1 KBrd + 1 mouse, but work KB only, while mouse doesn't, i had wait 5 min,  then i had used other wireless mouse to reconnect this mouse.
Of coarse, i had tried over GUI at first, but mouse didn't connected, while the signs of paired and trusted was in it's icon.

then i had used the CLI, and mouse was connected from the 1st attempt, i even not used pair and trust this time.

3. in spite of  at config file it was commented 2 devices, on the BTmanager panel there was all 4 devices icon, (and only one of them had green, red, and blue stripes, showing connection quality.) So may be have one more place where data keep, but this is not documented in installation manual on   github.com/blueman-project

5. if need some logs, i can to add, just suggest me, please, where they are placed on.

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2023-08-20:
`hcid.conf` is a thing of a long forgotten past.

If you encounter reproducible situations where you can pair or connect a device with `bluetoothctl` but the same step fails in blueman, please provide the blueman debug output of that attempt.

antondrroid, 2023-08-21:
if you can advice, where is output of blueman has stored in system, i can send it.  i had send dmesg already.

and if hcid.conf is absoleted, may be can you advice, which config file i need to use now?

cschramm, 2023-08-21:
See "Debugging blueman" on the troubleshooting page that you consulted earlier.

> and if hcid.conf is absoleted, may be can you advice, which config file i need to use now?

To do what?

antondrroid, 2023-09-10:
to comment items, which i want temporarily to disable

closed 2023-11-17 by github-actions[bot]