blueman-project/blueman #2121: Devices tray menu item should always bring up the dialog (like the other menu items)

chkothe, 2023-08-19:
If I click on `Devices...` in the tray menu I generally expect the window to be surfaced, since I am trying to interact with it. However, if the window is already open (but hidden or minimized), it will instead be _closed_. This differs from the behavior of the `Adapters...` and `Local Services...` menu entries and puts an unnecessary burden on the user.

The issue is that, if one wants to interact with Devices, one currently should first search their entire desktop(s) for an already open Devices window, and if that search comes up successful, surface it, and otherwise go to the tray menu and use the Devices entry to open a fresh window. Or alternatively click Devices, wait 1-2s for a window to open, and if none opens, assume that perhaps one was already open somewhere, click the button a second time, and wait whether it now opens or if perhaps the window manager or BT subsystem is in a bad state, preventing the window from opening. That's a much more complex user loop than "click on Devices and wait for it to open" and throws me off nearly 50% of the time that I click on it (which is frequent since my headphones disconnect in response to various events).

cschramm, 2023-08-20:
I agree that the behavior is a bit unexpected and unfavorable.

@infirit: I think you introduced that with https://github.com/blueman-project/blueman/commit/0e026d0cf497c5c80f46f6e0552fbc50814b07ff#diff-6d6c881deef28320434de26849f09e51819e7c2d0b4739ac60b2cc86562ea7fcR78. Was there a specific reason for the start / stop (Activate / Quit) behavior? :thinking:

infirit, 2023-08-20:
Not sure I understand what the problem is. Is it that blueman-manager opens behind other windows?

infirit, 2023-08-20:
> However, if the window is already open (but hidden or minimized), it will instead be _closed_.

If you mean this, then no this behaviour was how it was before which I replicated when moving to Gtk.Application. `kill(pid, 'blueman-manager')` specifically. It can be changed to always present the window.

cschramm, 2023-08-20:
Oh, sure, I missed the `kill` in the `if` condition. 🤦‍♂️

cschramm, 2023-08-20:
https://github.com/blueman-project/blueman/commit/d8bf0b89292a7573cf4f00c93891f1655f14029b actually explicitly introduced it. Not sure why it does not seem connected to a PR. 🤔

cschramm, 2023-08-20:
#315 is the source. Hm. 🤷‍♂️ I do not have a strong opinion how it should behave. I'd neither use the icon to close the window nor to bring the already open window up. It makes sense to bring it up though for cases where you do not realize that it's already open and expect to open it, so I'd go with the suggested change. We should probably check some other applets' current behavior to not deviate from the mass, though.

infirit, 2023-08-20:
The origin of the request is here #316. I can revert the behaviour, personally I don't care too much but the old behaviour is more consistent with the others.

cschramm, 2023-08-20:
Which one do you mean with old? 😅 The current or the pre-#316 one?

infirit, 2023-08-20:
Pre #316, don't close manager.

cschramm, 2023-08-20:
Perfect, then let's revert to that. No idea what the intention was to change it. Maybe other applications changed their behavior in the meantime and it was common back then. 🤷‍♂️

infirit, 2023-08-20:
#2122 will present the manager window instead of closing it. Whether it will be brought to the foreground is up to the desktop/windowmanager. At the very least it will be flashing/highlighted.

chkothe, 2023-08-20:
Thanks much for the quick resolution! Open source at its best

infirit, 2023-08-23:
#2122 was merged and will be in the next release.

closed 2023-08-23 by infirit