blueman-project/blueman #2126: Adapter Preferences popup window not closing

icedwater, 2023-08-23:
blueman: 2.1.2-1ubuntu0.3
BlueZ: 5.53-0ubuntu3.6
Distribution: Ubuntu 20.04.6 LTS
Desktop environment: i3

I had a similar issue to #2125 with the Adapters popup window not closing, though it might have a different cause.

Clicking Adapter -> Preferences spawns a popup window as a separate process (`/usr/bin/python3 /usr/bin/blueman-adapters`) and I can kill that without affecting the applet itself, but there is no way from within the dialog to close it, whether Ctrl-C, Ctrl-Q, Ctrl-W, or Esc.

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2023-08-23:
Accelerators got added in #1982.

icedwater, 2023-08-26:
Thanks! I guess I just need to build a more recent version then.

closed 2023-11-02 by github-actions[bot]