blueman-project/blueman #2133: how can i show the battery of my bluetooth device

zeyadelkholyttt, 2023-09-05:
blueman: latest
BlueZ: latest
Distribution: void
Desktop environment: kde

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

infirit, 2023-09-06:
Latest isn't very helpful. Regardless the latest stable release will show battery percentage in blueman-manager and in the connection notification. This however depends wether it's exposed over the BlueZ dbus api.

zeyadelkholyttt, 2023-09-07:
> Latest isn't very helpful. Regardless the latest stable release will show battery percentage in blueman-manager and in the connection notification. This however depends wether it's exposed over the BlueZ dbus api.

i found where i can get it in the panel but i will be very thx if i can get it in the kde bar

cschramm, 2023-09-07:
If upower picks up the device (check e.g. `upower -e`), desktops typically provide some kind of battery applet for it.

closed 2023-11-15 by github-actions[bot]