blueman-project/blueman #2134: Fedora 38 - Blueman fills log with messages

didierga, 2023-09-08:
blueman: blueman-2.3.5-3.fc38.x86_64 blueman-nautilus-2.3.5-3.fc38.noarch
BlueZ: bluez-5.69-1.fc38.x86_64
Distribution: Fedora 38 Workstation
Desktop environment: gnome-shell-44.4-1.fc38.x86_64

blueman fills syslog with these two massages:
```sept. 08 02:07:12 desktop nautilus[28067]: TypeError: BluemanSendtoExtension.get_background_items() missing 1 required positional argument: 'file'
sept. 08 02:07:12 desktop nautilus[28067]: TypeError: BluemanSendtoExtension.get_file_items() missing 1 required positional argument: 'files'```

cschramm, 2023-09-08:
nautilus 0.43 changes the extension interface. Can you test #2135? You could just apply the changes to your `/usr/share/nautilus-python/extensions/nautilus_blueman_sendto.py`.

didierga, 2023-09-08:
Fix in https://github.com/blueman-project/blueman/pull/2135 works fine.
Many thanks for tour prompt answer.

PR #2135 (2023-09-08): Adapt sendto extension for recent nautilus
nautilus 0.43 / nautilus-python 4.0 dropped the window argument:

https://gitlab.gnome.org/GNOME/nautilus/-/commit/bc55ff7599ef7f5acc6e9b648765fdb45b56431c

https://gitlab.gnome.org/GNOME/nautilus-python/-/commit/fac72de53ba7a102b22b64976bc49f68c97c23f7

Closes #2134
commit eb38dca8 (2023-09-08): Adapt sendto extension for recent nautilus

nautilus 0.43 / nautilus-python 4.0 dropped the window argument:

https://gitlab.gnome.org/GNOME/nautilus/-/commit/bc55ff7599ef7f5acc6e9b648765fdb45b56431c

https://gitlab.gnome.org/GNOME/nautilus-python/-/commit/fac72de53ba7a102b22b64976bc49f68c97c23f7

Closes #2134
closed 2023-09-08 by cschramm